nightscout/AndroidAPS #1377: No carbs in AAPS if phone restart during bolus started via wizard

robertrub, 2022-02-25:
I entered carbs in wizard and started bolus injection. During the bolus, I had to restart the phone. Once restarted, I actualized pod status to validate the sent bolus.

I can see the bolus but there is no trace of the carbs in AAPS.

[image]

[image]

[image]

I don't know if the logs are reset after reboot...

[image]

MilosKozak, 2022-02-25:
carbs are recorded only if bolus is delivered successfuly. it's for safety because people entered carbs for 2nd time when repeated bolus

closed 2022-02-25 by MilosKozak

robertrub, 2022-02-25:
@MilosKozak When the phone booted, the bolus was "unsure" in the DASH tab. When I actualized it, the fact that the bolus was injected totally was known to AAPS.
I don't know if it will be doable to find a logic that adds carbs if they are missing or to show a message "check carbs and bolus" after boot if AAPS stopped before the bolus complete (the DASH tab new that it hadn't received the "bolus injected ok" message).

MilosKozak, 2022-02-25:
carbs info is store donly in memory until bolus is finished