nightscout/AndroidAPS #1386: [AAPS3.0.0.1] Wear dont show forecast, although is it enabled in wear-settings

rICTx-T1D, 2022-02-26:
Hi community,
I have installed the new version 3.0.0.1 from master-branch and deploy it on my phone and wearOS-watch.
On my phone, I make some settings new, because it was not migrated, as described in FAQ.
Now I habe the problem, that my watch (Fossil Gen5) don't show forecast-line in all watch faces with graph/diagram. I going to wear settings on phone, disable forecast, wait 10 minutes and activate it again. No success. The regular and forecast-line is shown on phone, but the watch show only regular WITHOUT forecast.

Can anyone confirm this problem or give support to fix it?

Best regards
rICTx

rICTx-T1D, 2022-02-26:
3.0.0.1-dev-b doesn't  have the problem, how Robby is reported in discord.
[https://media.discordapp.net/attachments/629954570394533889/947217606308421662/screen.png](Picture)

Philoul, 2022-02-26:
I tested today with dev-b and it works
I'll test tomorrow with dev-c

Philoul, 2022-02-27:
I confirm it works with dev-c too
=> I made my test with an old version of Wear apk (build with 2.8 and not updated recently)

rICTx-T1D, 2022-02-27:
I can't understand, what's the ghost has doing in the night with my watch, during it charged on Power supply. But today, I open my eyes and the forecast was shown.

closed 2022-02-27 by rICTx-T1D