nightscout/AndroidAPS #1387: Minor visual bug : no graphic in profile settings

robertrub, 2022-02-26:
When I look at my profile, the graphs at the bottom is not visible (for insulin duration, basals etc).

[image]

I need to select the profile to edit in order to have the graphics be filled.

[image]

MilosKozak, 2022-02-26:
version?

robertrub, 2022-02-26:
[image]

MilosKozak, 2022-02-26:
fixed by https://github.com/nightscout/AndroidAPS/commit/bbe6248f81068742d7b27c924790c159b35438a5