nightscout/AndroidAPS #1388: Loop is not executing

margogoodfellow, 2022-02-26:
I had this problem before.  I have built new app from scratch and I still have this problem.  Loop works fine at first and then doesn't execute.  Bg is changing and loop doesn't respond for 10 -15 min. This happened lots of times today. Last one at 21:30

[image]

MilosKozak, 2022-02-28:
i need exact time(s) when this happened

margogoodfellow, 2022-02-28:
I don't understand why this happening as after last time I have rebuilt
completely and got new phone. Is this bug transferred with preferences or
is there something to do with nightscout?

margogoodfellow, 2022-03-03:
https://drive.google.com/file/d/1O4Ol1VJYPyAhxJ1fM_85rEJvErRkW8oz/view?usp=drivesdk
Not executing from 7:50 am till 8:05 am I had to press run now

MilosKozak, 2022-05-27:
ver 2.8 is unsupported now
test with 3.0

closed 2022-05-27 by MilosKozak