nightscout/AndroidAPS #1394: Entrys in the treatment menu only partially visible

cr582, 2022-02-28:
In the treatment menu, entries are only partially visible and can therefore no longer be deleted (at least in portrait mode). See linked screenshot.

[In the treatment menu, entries are only partially visible and can therefore no longer be deleted (at least in portrait mode). See linked screenshot.

[https://i.imgur.com/onA0pKx.jpg](https://i.imgur.com/onA0pKx.jpg)

MilosKozak, 2022-02-28:
already improved in dev branch

closed 2022-02-28 by MilosKozak

Andries-Smit, 2022-03-07:
@cr582 FYI fixed in PR #1362

cr582, 2022-07-20:
unfortunatelly it's even worse in 3.1.0
[image]

osodebailar, 2022-07-20:
It's all ok.
Click on the trashbin 😉

cr582, 2022-07-20:
even if I click on the trash icon and select something I am not able to see what is selected. Sorry, but this is not a solution to the mentioned issued and I really would appriciate it if someone could reopen this bug report since it is in no means resolved.

[image]

osodebailar, 2022-07-20:
Have you choose an own font and/or resolution size ? It looks like very big to me ?

cr582, 2022-07-20:
this is a screenshot with the smallest font setting available on Samsung Galaxy S20 FE
[image]
E

cr582, 2022-07-20:
and thsi is a screenshot of the start screen (with smallest font size) - I'm not able to see how many units of insulin are still available in the cartridge

[image]

osodebailar, 2022-07-20:
It's not a font issue. Maybe a Nightscout issue. Had the same problem a week ago .

cr582, 2022-07-20:
smartphone in landscape mode shows everything. I strongly doubt that this is a nighscout issue

[image]

cr582, 2022-07-21:
@osodebailar @MilosKozak should I open a new bug report for 3.1.0 or do you reopen this bug since it is clearly not fixed with #1362 
Thanks in advance

MilosKozak, 2022-07-21:
what phone are we talking about?

cr582, 2022-07-21:
first report: Huawei Mate 20; second report Samsung Galaxy S20 FE 5G