nightscout/AndroidAPS #1395: SMB bolus cancelled without alerts

Antoine-35, 2022-02-28:
Hello,

Version 3.0.1 dev branch 27/02/2022, with DynISF.
Pump Omnipod Dash
No timeout or communication problems with the pump
SMB always true

I noticed that for several hours I had no SMB. AAPS offered to inject some but it never happened.

SMB works well in master version 3.0.0 or in GitHub.com/tim2000s/AndroidAPS-2

I notice that the SMB is cancelled in the logs.

It could be interesting to have an alert in the interface in case of injection problem. Like during a timeout

```
15:30:16.029 Microbolusing 1.3U
15:30:16.575 [RxCachedThreadScheduler-28] D/PUMPQUEUE: [CommandQueueImplementation.add():149]: Adding: CommandSMBBolus - SMB BOLUS 1.30 U
...
15:30:16.580 [Thread-453] D/PUMPQUEUE: [QueueThread.run():113]: connect
15:30:16.581 [Thread-453] D/EVENTS: [RxBus.send():23]: Sending EventPumpStatusChanged[error=,performingAction=,secondsElapsed=0,status=CONNECTING]
15:30:16.582 [Thread-453] I/PUMP: [OmnipodDashPumpPlugin.connect():233]: connect reason=Connection needed
15:30:16.583 [Thread-453] D/EVENTS: [RxBus.send():23]: Sending EventOmnipodDashPumpValuesChanged[]
...
15:30:17.585 [Thread-453] D/PUMPQUEUE: [QueueThread.run():107]: connecting 1
15:30:17.586 [Thread-453] D/EVENTS: [RxBus.send():23]: Sending EventPumpStatusChanged[error=,performingAction=,secondsElapsed=1,status=CONNECTING]
15:30:18.587 [Thread-453] D/PUMPQUEUE: [QueueThread.run():107]: connecting 2
15:30:18.589 [Thread-453] D/EVENTS: [RxBus.send():23]: Sending EventPumpStatusChanged[error=,performingAction=,secondsElapsed=2,status=CONNECTING]
15:31:13.261 [RxCachedThreadScheduler-34] D/PUMP: [OmnipodDashManagerImpl$PodEventInterceptor.accept():667]: Intercepted PodEvent in OmnipodDashManagerImpl: Connected
15:31:13.359 [Thread-453] D/PUMPQUEUE: [QueueThread.run():122]: connection time 56s
15:31:13.364 [Thread-453] D/PUMPQUEUE: [QueueThread.run():128]: performing SMB BOLUS 1.30 U
15:31:13.365 [Thread-453] D/EVENTS: [RxBus.send():23]: Sending EventQueueChanged[]
15:31:13.369 [Thread-453] D/EVENTS: [RxBus.send():23]: Sending EventPumpStatusChanged[error=,performingAction=SMB BOLUS 1.30 U,secondsElapsed=0,status=PERFORMING]
15:31:13.698 [Thread-453] D/PUMPQUEUE: [CommandSMBBolus.execute():26]: Last bolus: 1646049702024 28/02/2022 13:01:42
15:31:13.704 [Thread-453] D/PUMPQUEUE: [CommandSMBBolus.execute():34]: SMB bolus canceled. deliverAt: 28/02/2022 15:29
15:31:13.706 [Thread-453] D/PUMPQUEUE: [CommandSMBBolus.execute():36]: Result success: false enacted: false
```

[AndroidAPS._2022-02-28_15-42-10_.log](https://github.com/nightscout/AndroidAPS/files/8156567/AndroidAPS._2022-02-28_15-42-10_.log)

Antoine-35, 2022-03-02:
Different version: GitHub.com/tim2000s/AndroidAPS-2

Same issue.
SMB was working well until then.

I see it because in the curve the basal went to 0 because of the SMB (which is not injected) then resumed at the maximum

```
10:31:03.145 [RxCachedThreadScheduler-4] D/APS: [DetermineBasalAdapterSMBJS.invoke():141]: Result: {"temp":"absolute","bg":187,"tick":"+15","eventualBG":217,"targetBG":100,"insulinReq":1.96,"deliverAt":"2022-03-02T09:30:46.063Z","sensitivityRatio":1,"predBGs":{"IOB":[187,185,183,180,178,175,172,168,165,161,157,152,148,143,139,135,132,129,126,123,120,118,115,113,111,110,108,107,105,104,103,102,101,100,99,98,98,97,97,96,96,96,95,95,95,94],"ZT":[187,178,170,162,154,147,141,134,128,123,117,112,108,103,99,95,92,88,85,83,80,77,75,73,71,69,68,66,65,64,63,62,61,60,59,58,58,57,57,56,56,55,55,55,54,54,54,54],"COB":[187,185,184,182,181,180,179,179,178,177,177,177,176,176,176,176,176,176,175,175,175,175,175,174,174,173,173,172,171,170,169,168,167,166,166,165,164,164,163,163,162,162,162,161],"UAM":[187,198,207,216,224,232,238,244,249,252,255,258,259,259,259,257,255,251,248,246,243,240,238,236,234,232,231,229,228,227,226,225,224,223,222,221,221,220,220,219,219,218,218,218,217]},"reason":"COB: 37.9, Dev: 123, BGI: -9, ISF: 31, CR: 10.5, Target: 100, minPredBG 160, minGuardBG 185, IOBpredBG 94, COBpredBG 161, UAMpredBG 217; Eventual BG 217 >= 105,  insulinReq 1.96; setting 60m low temp of 0U/h. Microbolusing 0.95U. ","COB":37.85041666666665,"IOB":4.228,"units":0.95,"rate":0,"duration":60}
10:31:03.157 [RxCachedThreadScheduler-4] D/APS: [Profiler.log():15]: >>> SMB calculation <<< executed in 17256 milliseconds
10:31:03.159 [RxCachedThreadScheduler-4] D/EVENTS: [RxBus.send():23]: Sending EventOpenAPSUpdateGui[]
...
10:31:50.842 [Thread-382] D/PUMPQUEUE: [CommandSMBBolus.execute():34]: SMB bolus canceled. deliverAt: 02/03/2022 10:30
10:31:50.844 [Thread-382] D/PUMPQUEUE: [CommandSMBBolus.execute():36]: Result success: false enacted: false
```

Antoine-35, 2022-03-02:
So it seems to be random. As one SMB was just given with success

Antoine-35, 2022-03-06:
I came back to OpenapsSMB, and the problem seems to have disappeared.

So this is related to DynamicISF, maybe a timeout problem because the SMB calculation is longer, close to a minute for my entry level phone.
But there is no information about the reason of the bolus cancellation in the logs

Antoine-35, 2022-03-06:
Works well in OpenAPS SMB so it is related to DynISF.
Maybe the SMB calculation: 60000 to 100 000ms in DynISF vs 1000ms in OpenAPS SMB

kamlox, 2022-03-26:
The problem seems to be the calculation of the TDD, because it is also very long in the statistics page.
Maybe modify the database to calculate the day once it is finished

kamlox, 2022-06-13:
I tried with the last dev, with the cache TDD. Everything is smoother but same issue

kamlox, 2022-06-13:
[image]

kamlox, 2022-06-13:
[image]

MilosKozak, 2022-06-13:
it's because min distance between boluses

kamlox, 2022-06-13:
I don’t think so as no blouses where enacted for more than an hour.

They were suggested in the script (and not limited by the minimum duration between two injections) but no injections.
They are cancelled without information

MilosKozak, 2022-06-13:
sorry right, canceled because the execution (start of delivery) took longer than 3 min

kamlox, 2022-06-13:
Any ideas to found what is the root cause of this delay?

MilosKozak, 2022-06-13:
first example in this issue is combination of:
* run just after app start when phone is still very busy
* 1 min to connect to pump

kamlox, 2022-06-13:
Ok, thanks for the explanation 👍🏻.
It may be a problem between my cheap phone and the pump. It must take less than a minute to connect.

kamlox, 2022-06-16:
What is strange it’s I have a little bit this issue also on non dynISF version on the dev but not on the master.

closed 2023-03-27 by antoinekh