nightscout/AndroidAPS #1398: Loop suspended after decoding the wrong pump history.

diyaps, 2022-03-01:
Reporting bugs
03-01 13:03:50.491 D 26980    27103    DATABASE:                                                   [Thread-12]: [PumpSyncImplementation.createOrUpdateTotalDailyDose():431]: Updated TotalDailyDose TotalDailyDose(id=47, version=1, dateCreated=1646111030486, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=MEDTRONIC_523_723_REVEL, pumpSerial=573079, temporaryId=null, pumpId=20220301235959110, startId=null, endId=null), timestamp=1646150399000, utcOffset=28800000, basalAmount=1.2, bolusAmount=3.0, totalAmount=4.2, carbs=0.0) 
03-01 13:03:50.492 D 26980    27103    PUMP                                                           : [Thread-12]: [MedtronicHistoryData.processNewHistoryData():345]: ProcessHistoryData: Bolus [count=1, items=[{"entryType":"Bolus","DT":"01.03.2022 12:08:55","atechDateTime":20220301120855,"decodedData":{"Object":{"bolusType":"Normal","deliveredAmount":1.5,"duration":0,"requestedAmount":1.5}}}]] 
03-01 13:03:50.492 D 26980    27103    PUMP                                                           : [Thread-12]: [MedtronicHistoryData.processBolusEntries():507]: DD: entryWithTempId=null 
03-01 13:03:50.496 D 26980    27103    PUMP                                                           : [Thread-12]: [MedtronicHistoryData.processBolusEntries():542]: syncBolusWithPumpId [date=20220301120855, pumpId=20220301120855001, insulin=1.50, pumpSerial=573079] - Result: false 
03-01 13:03:50.496 D 26980    27103    PUMP                                                           : [Thread-12]: [MedtronicHistoryData.processNewHistoryData():357]: ProcessHistoryData: TBRs Processed [count=0, items=[]] 
03-01 13:03:50.497 D 26980    27103    PUMP                                                           : [Thread-12]: [MedtronicHistoryData.getSuspendResumeRecordsList():922]: SuspendResume Records: [PumpHistoryEntry [type=SuspendPump          [30 , 0x1E], DT: 30.00.2060 00:20:60, length=2,5,0(7), data=No data]] 
03-01 13:03:50.497 D 26980    27103    PUMP                                                           : [Thread-12]: [MedtronicHistoryData.getNoDeliveryRewindPrimeRecordsList():982]: Prime Records: [PumpHistoryEntry [type=Prime                [3  , 0x03], DT: 01.03.2022 12:55:24, length=5,5,0(10), head=00 00 00 00, datetime=18 F7 2C 01 16, rawData=03 00 00 00 00 18 F7 2C 01 16]] 
03-01 13:03:50.498 D 26980    27103    PUMP                                                           : [Thread-12]: [MedtronicHistoryData.getNoDeliveryRewindPrimeRecordsList():994]: Filtered Records: [PumpHistoryEntry [type=NoDeliveryAlarm      [6  , 0x06], DT: 01.03.2022 12:55:58, length=4,5,0(9), data=], PumpHistoryEntry [type=Prime                [3  , 0x03], DT: 01.03.2022 12:55:24, length=5,5,0(10), head=00 00 00 00, datetime=18 F7 2C 01 16, rawData=03 00 00 00 00 18 F7 2C 01 16], PumpHistoryEntry [type=Rewind               [33 , 0x21], DT: 01.03.2022 12:54:32, length=2,5,0(7), data=No data], PumpHistoryEntry [type=NoDeliveryAlarm      [6  , 0x06], DT: 01.03.2022 12:52:26, length=4,5,0(9), data=], PumpHistoryEntry [type=Bolus                [1  , 0x01], DT: 01.03.2022 12:08:55, length=8,5,0(13), head=00 3C 00 3C 00 0E 00, datetime=37 C8 2C 41 16, rawData=01 00 3C 00 3C 00 0E 00 37 C8 2C 41 16]] 
03-01 13:03:50.498 D 26980    27103    PUMP                                                           : [Thread-12]: [MedtronicHistoryData.showLogs():107]: NoDeliveryRewindPrimeRecords: Records to evaluate:  
03-01 13:03:50.498 D 26980    27103    PUMP                                                           : [Thread-12]: [MedtronicHistoryData.showLogs():111]: [{"entryType":"Prime","DT":"01.03.2022 12:55:24","atechDateTime":20220301125524,"decodedData":{"Amount":0.0,"FixedAmount":0.0}},{"entryType":"Rewind","DT":"01.03.2022 12:54:32","atechDateTime":20220301125432,"decodedData":{}},{"entryType":"NoDeliveryAlarm","DT":"01.03.2022 12:52:26","atechDateTime":20220301125226,"decodedData":{}}] 
03-01 13:03:50.501 D 26980    27103    PUMP                                                           : [Thread-12]: [MedtronicHistoryData.processNewHistoryData():374]: ProcessHistoryData: Delivery Suspend Processed [count=1, items=[{}]] 
03-01 13:03:50.505 D 26980    27103    PUMP                                                           : [Thread-12]: [MedtronicHistoryData.processSuspends():886]: processSuspends::syncTemporaryBasalWithPumpId [date=20220301125226, rate=0.0, duration=178 s, type=PUMP_SUSPEND, pumpId=20220301125226006, pumpSerial=573079] 
03-01 13:03:50.513 D 26980    27103    PUMP                                                           : [Thread-12]: [MedtronicHistoryData.processSuspends():902]: syncTemporaryBasalWithPumpId: Result: false 
03-01 13:03:50.514 E 26980    27103    CORE                                                           : [Thread-12]: [MedtronicHistoryData.finalizeNewHistoryRecords():204]: Problem decoding date from last record: PumpHistoryEntry [type=SuspendPump          [30 , 0x1E], DT: 30.00.2060 00:20:60, length=2,5,0(7), data=No data] 
03-01 13:03:50.514 E 26980    27103    CORE                                                           : [Thread-12]: [MedtronicHistoryData.finalizeNewHistoryRecords():222]: Since we couldnt determine date, we dont clean full history. This is just workaround. 
03-01 13:03:50.514 I 26980    27103    PUMP                                                           : [Thread-12]: [MedtronicUIComm.executeCommand():29]: Execute Command: GetRemainingInsulin 
03-01 13:03:50.514 D 26980    27103    PUMP                                                           : [Thread-12]: [MedtronicUITask.execute():53]: MedtronicUITask: @@@ In execute. GetRemainingInsulin 
03-01 13:03:50.514 D 26980    27103    PUMPCOMM:                                                   [Thread-12]: [MedtronicCommunicationManager.getRemainingInsulin():711]: getDataFromPump: GetRemainingInsulin 
03-01 13:03:50.515 I 26980    27103    PUMPBTCOMM:                                                 [Thread-12]: [RileyLinkCommunicationManager.sendAndListen():77]: Sent:A7 57 30 79 73 00 
03-01 13:03:50.515 D 26980    26980    PUMP                                                           : [main]: [MedtronicFragment.onResume$lambda-8():142]: onStatusEvent(EventRileyLinkDeviceStatusChangeEventRileyLinkDeviceStatusChange[errorDescription=<null>,pumpDeviceState=Active,rileyLinkError=<null>,rileyLinkServiceState=<null>,rileyLinkTargetDevice=<null>] 
03-01 13:03:50.516 D 26980    26980    PUMP                                                           : [main]: [MedtronicFragment.setDeviceStatus():227]: Command: GetRemainingInsulin 

--------------

I use AAPS 3.0.0, After reset the medtronic 723 pump and connect again, it only can loop for less than 5min and then the AAPS suspends. But I checked my pump, it doesn't suspend.
I have checked the log and found the errors above. In addition, I found an abnormal timestamp in pump history. It shows "30.00.2060 00:20:60 Suspend"

andyrozman, 2022-03-01:
@diyaps Need full log here, please upload it. There seems to be some problem in the history parsing so I need to see what happened before.

diyaps, 2022-03-01:
[logcat-0301-020618.log](https://github.com/nightscout/AndroidAPS/files/8163034/logcat-0301-020618.log)
Hi @andyrozman log file is attached. I found there is crash when reading the history

andyrozman, 2022-03-04:
I need the real logs... For some reason, your logs are missing the whole history page... I have no idea why? 
We are logging out whole history, but your logcat logs don't have it...

aapsLogger.info(LTag.PUMPBTCOMM, "History Page Data:\n$sb")

insulinplayer, 2022-03-04:
@andyrozman  I'm not sure what happened to my logcat... But fortunately, I saved another log file including "History Page Data". Here is it
[logcat-0301-023655.log](https://github.com/nightscout/AndroidAPS/files/8189306/logcat-0301-023655.log)
.

andyrozman, 2022-03-05:
This history looks ok... I run it through decoder... Don't you have logs from when this error happened? They should be on your phone.

andyrozman, 2022-03-05:
I am looking through logcat of first file you sent and it looks like data didn't come through correctly, it seems like there were some timeouts durring communication... Really weird... Maybe some problem with RL or RL-clone you are using...

andyrozman, 2022-06-25:
@diyaps Is this fixed for you with new dev?

diyaps, 2022-10-18:
@andyrozman  I'm currently using 3.1.0.2 on master, not fixed

diyaps, 2022-10-18:
[AndroidAPS._2022-10-17_18-16-45_.0.zip](https://github.com/nightscout/AndroidAPS/files/9810807/AndroidAPS._2022-10-17_18-16-45_.0.zip)
Not sure this log will belp or not. 
I didn't see how and when the wrong suspend event was written to the pump history.
This log may only contain the reading pump history and decoding part.

MilosKozak, 2026-10-15:
Found the cause in the history page from the log of 2022-10-17.

After the pump reset, the 723 wrote some sensor setting records into its history (0x50, 0x54, 0x55). AAPS read three of them with the wrong length:
- 0x55 is 55 bytes, AAPS read 7
- 0x54 is 64 bytes, AAPS read 7
- 0x50 is 37 bytes on a 523/723, AAPS read 41 (41 is only for pumps with low suspend, the 554/754)

So the rest of these records was read as more records. One of them was a suspend with the date 30.00.2060 00:20:60. It was the newest entry in the history, so AAPS showed the pump as suspended all the time.

Fixed in 815a689c8f:
- The three records now have the right length, the same as in MinimedKit (the Loop driver). The page from your log now decodes without any wrong record.
- A record with a date that does not exist (for example month 0 or 60 seconds) is not used any more. So a record read from the wrong place cannot show the pump as suspended again.

Closing. If you still see this on a build with this fix, please open a new issue and attach the logs.

closed 2026-10-15 by MilosKozak