nightscout/AndroidAPS #1399: 3.0.0.1-dev-d crash

elgrupo123, 2022-03-01:
keeps crashing when the TBR could not be set.
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8160526/AndroidAPS.log)

MilosKozak, 2022-03-01:
https://console.firebase.google.com/u/0/project/androidaps-c34f8/crashlytics/app/android:info.nightscout.androidaps/issues/48657b9c03c2a5e1f189fa6c6d5fc9ac?versions=3.0.0.1-dev-d%20(1500)&time=last-seven-days

elgrupo123, 2022-03-01:
> https://console.firebase.google.com/u/0/project/androidaps-c34f8/crashlytics/app/android:info.nightscout.androidaps/issues/48657b9c03c2a5e1f189fa6c6d5fc9ac?versions=3.0.0.1-dev-d%20(1500)&time=last-seven-days

"This project does not exist or you do not have permission to view it".

Andries-Smit, 2022-03-04:
For those who can not access the Crashlytics, this is the log:
```
Fatal Exception: java.lang.IllegalStateException: Received message while trying to send
       at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.message.MessageIO.sendMessage(MessageIO.kt:42)
       at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.session.SessionEstablisher.negotiateSessionKeys(SessionEstablisher.kt:42)
       at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.session.Connection.establishSession(Connection.kt:167)
       at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.OmnipodDashBleManagerImpl.establishSession(OmnipodDashBleManagerImpl.kt:158)
       at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.OmnipodDashBleManagerImpl.connect$lambda-1(OmnipodDashBleManagerImpl.kt:139)
       at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.OmnipodDashBleManagerImpl.$r8$lambda$5TrAjNnaoab-abmIF0Xgs08Vojo()
       at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.OmnipodDashBleManagerImpl$$ExternalSyntheticLambda1.subscribe(:4)
       at io.reactivex.rxjava3.internal.operators.observable.ObservableCreate.subscribeActual(ObservableCreate.java:41)
       at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13176)
       at io.reactivex.rxjava3.internal.operators.observable.ObservableDoOnEach.subscribeActual(ObservableDoOnEach.java:42)
       at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13176)
       at io.reactivex.rxjava3.internal.operators.observable.ObservableDefer.subscribeActual(ObservableDefer.java:40)
       at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13176)
       at io.reactivex.rxjava3.internal.operators.observable.ObservableDoOnEach.subscribeActual(ObservableDoOnEach.java:42)
       at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13176)
       at io.reactivex.rxjava3.internal.operators.observable.ObservableDoOnEach.subscribeActual(ObservableDoOnEach.java:42)
       at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13176)
       at io.reactivex.rxjava3.internal.operators.observable.ObservableSubscribeOn$SubscribeTask.run(ObservableSubscribeOn.java:96)
       at io.reactivex.rxjava3.core.Scheduler$DisposeTask.run(Scheduler.java:644)
       at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.run(ScheduledRunnable.java:65)
       at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.call(ScheduledRunnable.java:56)
       at java.util.concurrent.FutureTask.run(FutureTask.java:266)
       at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
       at java.lang.Thread.run(Thread.java:920)
```

mick711, 2022-03-06:
@
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8192137/AndroidAPS.log)
Crash 8h31 this morning with dev d, Android 12 Zenfone 8 last update yesterday before dev d solve this  problem.

mick711, 2022-03-06:
Dev c too  
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8193584/AndroidAPS.log)
23h43

elgrupo123, 2022-03-07:
@Andries-Smit works just fine!  Thanks!!!

closed 2022-03-14 by MilosKozak