nightscout/AndroidAPS #1401: [3.0.0.1] After entering carbohydrates in the bolus calculator, no carbohydrates are displayed.

69nobody, 2022-03-01:
I have noticed several times now that when using the bolus calculator, after entering the carbohydrates and confirming with OK, no carbohydrates are displayed on the main screen. I have attached logs. Time approx. 14:30
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8162019/AndroidAPS.log)

mick711, 2022-03-01:
Me too but not all the time, sometimes it work.

69nobody, 2022-03-02:
I do not have other views/screenshots

[image]
[image]

MilosKozak, 2022-05-27:
no data in log. create new issue with more screenshots and log if you can replicate

closed 2022-05-27 by MilosKozak