nightscout/AndroidAPS #1404: 3.0.0.1-dev-d crash / closes automatically when sliding between tabs and reach Combo pump tab

parapenT1sta, 2022-03-03:
Crash between 2:43 and 2:44AM

[AndroidAPS_LOG_1646275450352.log.zip](https://github.com/nightscout/AndroidAPS/files/8174523/AndroidAPS_LOG_1646275450352.log.zip)

parapenT1sta, 2022-03-03:
The video:

https://user-images.githubusercontent.com/608650/156487458-ab4c8669-5be3-4f47-9893-5a8275879781.mp4

parapenT1sta, 2022-03-13:
@MilosKozak Do you have any other report about this?

Andries-Smit, 2022-03-15:
@parapenT1sta can you change in your preferences the `Show comm. error count` settings, and try again?

parapenT1sta, 2022-03-16:
@Andries-Smit The fact is that none of the option was selected (empty) now I choose "Always" and I have no crash... so far. Maybe a value must be set but to be honest I didn't see this option before and that's why I didn't select. Is it possible to make a default option to avoid this issue in the future... if in fact this was the issue.

MilosKozak, 2022-03-16:
turn off autosens logging until you have very good reason to do it. it makes logs unreadable

Andries-Smit, 2022-03-16:
@parapenT1sta just set it to `never`. 
There is a default in the code, so normally you should not have any issue.  In your case, an incorrect value setting was stored. The question was, how did the setting get into the storage, will answer the question if others will be affected. It might be that only people on the dev branch will have this issue, though that should not happen, I expect the people the move from release to release will not be affected.

parapenT1sta, 2022-03-16:
@Andries-Smit I just want to thank you for the support. If wasn't you I would never find the issue! :-) Now it's solved for me and hopefully I'm the only one affected.

I'm closing this issue.

closed 2022-03-16 by MilosKozak