nightscout/AndroidAPS #1410: After G6 CGM is deaktivated, TT was still on 0% and must be canceled manually.

Alamo04, 2022-03-04:
My Dexcom Sensor Failed, shows 47 and my bg was at 110. So I stopped the sensor.
After 51 minutes without values, AAPS didn't cancel his zero temp, so I must do that manually. 
The bg was rising up to 210 in that time.

yhonatanb20, 2022-03-20:
Experiencing same issue, starnge that the algorithm doesn't suspend itself when there are no BG values.

philipgo, 2022-03-26:
IIRC zero temps should be limited to 30 minutes when blood glucose values are missing: https://github.com/openaps/oref0/blob/acb0150021e7bf284b6741a9510076a13f690747/lib/determine-basal/determine-basal.js#L161

jotomo, 2022-04-21:
OpenAPS is only invoked when a new glucose value arrives and can manually be invoked only for glucose values max 7 minutes old. Hence the code mentioned above isn't invoked and the active temp runs until finished (I'm guessing in an OpenAPS setup, the algo is triggered periodically, so it runs even if no new glucose values have arrived?)
Given the last inputs to OpenAPS in this case a long zero temp seems like an appropriate defensive action.
With a failing sensor like that the user needs to take over control and take appropriate action I'd say.

closed 2022-04-21 by Alamo04