nightscout/AndroidAPS #1411: 25u bolus given without my knowledge, which bypassed hard limits

benedicthenry, 2022-03-05:
My hard settings in AAPS are: 'Adult' patient with a maximum bolus of 17.5U; Max total IOB is 22.17; max minutes of basal to limit smb to is 35 (mins). Despite these hard limits, I received a bolus of 25U yesterday morning at 11.09, which I wasn't aware of and which I didn't enter into AAPS that I'm aware of, leading to a day of recurrent, scary hypo's. I've attached the log from yesterday. I'd be grateful for some help to try to figure out how this happened please, because it was dangerous and frightening and I'd hate for it to happen again. My Insight pump only administers instant boluses in 0.1U increments and vibrates each time and so it's inconceivable and I would have almost certainly noticed if this had been what led to the bolus.

My set-up is: Pixel 4a; AAPS 3.0.0.1; Accu-Chek Insight; Libre 2; OOP2; xDrip+

[AndroidAPS._2022-03-04_17-50-59_.pdf](https://github.com/nightscout/AndroidAPS/files/8190499/AndroidAPS._2022-03-04_17-50-59_.pdf)

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2022-03-16:
there is something wrong with the pump. whole day repeats "Pump started" and  "Pump paused"
i see bolus
```
11:20:18.258 [Thread-89556] D/DATABASE: [PumpSyncImplementation.syncBolusWithPumpId():191]: Inserted Bolus Bolus(id=625, version=0, dateCreated=1646392818253, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=ACCU_CHEK_INSIGHT_BLUETOOTH, pumpSerial=31024263, temporaryId=null, pumpId=8623, startId=null, endId=null), timestamp=1646392145248, utcOffset=0, 
 mount=25.0, type=NORMAL, isBasalInsulin=false, insulinConfiguration=null)
```
but not entered through AAPS (probably on pump)

based on this didn't you press buttons on pump randomly?

closed 2022-03-16 by MilosKozak

benedicthenry, 2022-03-16:
Thanks Milos. I didn't knowingly press any buttons on the pump, it's just a relief to know that it wasn't AAPS. Thanks again and thank you for all your other work with AAPS.