nightscout/AndroidAPS #1420: [AAPS 3.0.0.1] Omnipod Dash Priming Bugs

JazzZ1, 2022-03-08:
Time of occurance in log: 08.03.2022, 11:15
Description:

###[Bug 1]###
Now I have my fourth Dash in row connected and working totally fine. But after priming and setting the canulla, it always shows a connection error. So i just have to click retry and it works fine. But this happens every single time on exactly these 2 steps, so I guess this could be a bug, simply fixed by a automated reconnect at the end of priming and setting.

Pod Firmware: 4.10.0

So here are my screenshots:

[image]
[image]

So after priming I always get this error, where I every time need to click retry to be able to continue:
[image]
[image]
[image]
[image]

And after continuing to attach, it immediatly shows the same error again, so again I need to click retry to be able to attach the pump:
[image]
[image]
[image]

###[Bug 2]###
I use the setting "Display always on", but still the display turns off while priming and attaching the Dash pump. On all other screens it works normally.

Log: 
[AndroidAPS_pump.log](https://github.com/nightscout/AndroidAPS/files/8207928/AndroidAPS_pump.log)

Build: 479bfc4-2022.02.06-18:41

macarolobe, 2022-03-13:
Same errors for me

keltakmaster, 2022-05-18:
Yeah, same issue here too!

Mynez, 2022-05-30:
Not work for me, it does not initialised

closed 2022-06-19 by MilosKozak