nightscout/AndroidAPS #1421: G6 sensor change not correctly sent to NS. Not sure if is an AAPS bug...

robertrub, 2022-03-08:
I use BYODA and G6 sensor. I _Stop sensor_ and then _Start new sensor_ in BYODA. This information arrives and is treated correctly in AAPS. I can see sensor age in AAPS.

The bug is (maybe) in the way this information (new sensor insertion/start) is sent to NS by AAPS. The sensor inserted message is on the treatments tab of NS but NS doesn't show SAGE data.

[image]

People seem to have the SAGE working in NS if they note the sensor change **in AAPS**. In my case, the sensor change information arrives from BYODA to AAPS.

[image]

robertrub, 2022-03-08:
I might have found the culprit. This had happened before. The CGM change date is totallt wrong in treatments in AAPS...

[image]

And here is the real sensor change date from BYODA.

[image]

Sorry, no logs :(

robertrub, 2022-03-08:
It is becoming more and more strange...

[image]

AAPS has the real insertion date too. I don't know from when the 2106 entry dates...

I can't delete the 2106 entry, it says _invalid_ after I tried to delete it...

CaptainBalou, 2022-03-08:
Same entry for the same date here... I can't see the year but because it is on the top of the list I assume it is in the future and so it would be the same.

[image]
[image]

robertrub, 2022-03-08:
@CaptainBalou Funny, it is the same exact date as mine... Even the same time stamp !
It is not something that came from BYODA most probably but where did AAPS get this entry is a mystery...

CaptainBalou, 2022-03-08:
@robertrub Yes, just because it is a full match I decided to write that comment. :-) I also use BYODA. Unfortunately I have no suggestions what the issue could be. Just wante to tell that you're not alone with this. ;-)

kristinsmotz, 2022-03-10:
I have seen this as well a few weeks ago, definitely year 2106, don't remember exact date. Also resolved it by deleting future treatments in Nightscout Admin tools.

happygirlnyc70, 2022-03-12:
I also had the future entry.  I removed it from the treatment tab.  Nightscout resolved itself at my next sensor change. Using dev-e.

closed 2022-03-16 by MilosKozak