nightscout/AndroidAPS #1425: Minor esthetic bug : Black on gray text

robertrub, 2022-03-09:
This "bug" was solved for carb entries but is still in preferences.

There are black "buttons" on gray background in the preference settings. It is very classy and chic but not the most visible combination.

[image]

mick711, 2022-03-09:
[image]

Same with Filtre on the top

Andries-Smit, 2022-03-10:
The text color on buttons are based on `colorPrimary`. The night theming, is not using the default style correctly. This requires a bit of rework. There are some plans the cleanup and consolidate styling.

osodebailar, 2022-03-10:
Its solved in the Themeswitcher branch for the light and dark mode. 
There must be planned to merge it in the future and apps has a real material theme and so we can change and adapt Color's as we want

Philoul, 2022-03-11:
Hi Andreas,

Are number picker fixed in your light/dark mode PRs (#1430 and/or #436) ?
(Horizontal is a bit better than vertical number, but none of bath are "perfect" currently)
[image]

osodebailar, 2022-03-11:
In my https://github.com/nightscout/AndroidAPS/pull/326 they look 
[image]
[image]

Philoul, 2022-03-19:
It's better on latest version but can be improved for numberpicker I think:
"-" is too small in horizontal number picker
"+" is not vertical centered (vertical and horizontal numberpickers), the bottom is troncated (vertical numberpicker) and has not same size (horizontal + bigger than vertical +) ...

[image]

osodebailar, 2022-03-19:
> It's better on latest version but can be improved for numberpicker I think: "-" is too small in horizontal number picker "+" is not vertical centered (vertical and horizontal numberpickers), the [...]

The only way i see is that we try to replace + and - with svg graphics. maybe we have the right ones in fontawsome ?

Philoul, 2022-03-19:
On latest dev-g, the header of pugin in hamburger menu is no more visible (#1460), could it be a regression due to #1458?

osodebailar, 2022-03-19:
> On latest dev-g, the header of plugin in hamburger menu is no more visible (#1460), could it be a regression due to #1458?

solved in https://github.com/nightscout/AndroidAPS/pull/1463

MilosKozak, 2022-03-19:
@osodebailar ProfileHelper should be `AppTheme_NoActionBar` but this still has wrong colors

osodebailar, 2022-03-19:
> 

i will have a look

robertrub, 2022-03-19:
On the latest build tonight, the minus seems tiny and the plus is too low on my phone (S9 Android 10)

[image]

[image]

kristinsmotz, 2022-03-19:
> 

Same here on Huawei P30 Pro (And in NSClient on Samsung Galaxy ZFlip3). Built just now 30 minutes ago. 
[image]
[image]

Philoul, 2022-03-20:
And some regression on other ok/cancel popup:
[image]
[image]

parapenT1sta, 2022-03-20:
Black text is also present in Treatments option after introducing the value for insulin or carbs.

parapenT1sta, 2022-03-20:
[image]

parapenT1sta, 2022-03-20:
[image]

osodebailar, 2022-03-20:
No other screenshots needed .
The reason is in all situation the same .
( Mainactivity is not in the „real dark Mode „ )
Solve ist when I am back at home today evening or tomorrow morning.

closed 2022-04-07 by MilosKozak