nightscout/AndroidAPS #1426: Incorrect (future) timestamp in AAPS NightScout on Dex sensor change (BYODA)

AISMap, 2022-03-10:
Hi,

I use BOYDA for managing my Dex G6 sensor and AAPS 3.0.0.1 as receiver. Sensor changes are kept automatically and sent to NightScout. 

Since I changed to AAPS 3.0.0.1 the timestamp for sensor change is not correct:
[image]

sensor change happened 2022-Mar-10 07:26 CET - displayed in AAPS as well as sent timestamp to NightScout is `2106-02-07T06:28:1500Z` instead.
[image]

AAPS does not allow to delete this wrong data anymore, it is possible only flagging it as "invalid".

Please get that fixed as it spoils up NightScout data as well as AAPS. Thank you

reijoahola-nightscout, 2022-03-10:
This seems to be about the same problem: https://github.com/nightscout/AndroidAPS/issues/1421

ideaweb, 2022-03-11:
Quick question, did you use the action button to let APS know that you changed the Sensor? I had the same problem, when using the action button that date appears in future. But after avoiding the button its working for me because it seems that BYODA notifies APS automatically about sensor change. Maybe this is the trouble maker that APS has 2 dates... but of course maybe I am wrong.

CaptainBalou, 2022-03-12:
@ideaweb I used xdrip for a while and there I used the action button to reset the time of the sensor. Since a few months I'm back on BYOD app and do not need this anymore because of the automatic notification as you mentioned.

AISMap, 2022-03-12:
Hi,

No - I haven't done anything manually. AAPS creates the change date automatically after having received the info from BYODA. I am running this setup quite a while and with 2.8.2 there was never an issue.

/Arne

MilosKozak, 2022-03-16:
added validation https://github.com/nightscout/AndroidAPS/commit/58b49a05eaefd96a9273942b46217e0fe0fb71e1

closed 2022-03-16 by MilosKozak