nightscout/AndroidAPS #1441: [3.0.0.1] Buttons disappear when pump is disconnected

Rhabarbara745, 2022-03-13:
Since 3.0.0.1 it's not possible to access Insulin Dialog and BolusWizard when the pump is disconnected.

This leads to the following problems:

- it is not possible to tell if a bolus should be given when the pump is disconnected from the body (eg. when swimming)
- "Do no bolus, record only" checkbox is more or less useless now
(because the pump must be reconnected to enter that insulin -> SMB kick in at the moment of reconnection, before even having the chance to enter already injected insulin.)

Since this is a major drawback compared to previous versions, my proposal would be reverting to the old behavior and if necessary, disable the "OK" buttons in BolusWizard and InsulinDialog when disconnected.

MilosKozak, 2022-03-13:
this settings was in AAPS all the time. The reason is very simple. If pump is not initialized (not known to AAPS) it's not possible apply constrains from pump.
SMB is not kicked on reconection but on next BG

closed 2022-03-13 by MilosKozak

Rhabarbara745, 2022-03-13:
Thanks for your quick reply @MilosKozak 

I'm not sure if we're talking about the same thing. By "disconnecting" I meant pressing the disconnect buttons (15min, 30min, 1h, ...) in the loop dialog. (the pump is already known by aaps at this stage)

<p float="left">
[image]
[image]
</p>

MilosKozak, 2022-03-14:
undestand now. but the answer is still the same. it can be confusing to see insulin dialog when pump is disconnected