nightscout/AndroidAPS #1451: Watch Face AAPSv2 Missing IOB and 10min BG Delta

erobinson, 2022-03-17:
* Running latest dev for AndroidAPS app & wear app. 
* WatchFace AAPSv2 but also seems to impact other watch faces
* Running w/ Dyanmic ISF config (not sure if this makes a difference

[image]
)

@Andries-Smit

Andries-Smit, 2022-03-19:
@erobinson is this still happening?  I think it should be temporary. Not all data is available at the start of your watch app. Therefore the data will be updated with 3 different data sets.

erobinson, 2022-03-20:
It's been like that for a few weeks, haven't done any debugging/testing to isolate the problem.

erobinson, 2022-03-23:
Found that I didn't have the show delta & show IOB settings turned on 😢 All working fine.

closed 2022-03-23 by erobinson