nightscout/AndroidAPS #1466: Wear OS bug aaps 3.0.0.1

yhonatanb20, 2022-03-20:
Hello, wanted to report a bug regardind the watch app (for wear OS): The watch face displays only readings that it received at real time, which means:
1. If you set a watch face, you should press "resend all data" on wear tab of phone app to see your graph for past few hours.
2. If my watch disconnects from the phone for 30 min at night, I then wake up and see a 30 min gap in the redings which doesn't really exist.
Can it be fixed please? Xdrip watchfaces, for example, become synced instantly with the phone app and show all data within a few seconds after you set them up.

Andries-Smit, 2022-03-20:
@yhonatanb20 thank you for reporting, there are many improvements on the wear os app now available on de dev branch. Including fixes for the issue you describe. If possible you can test it with the dev branch by updating just the wear os app. Please note that the quick wizard tiles will only work when both phone and watch are on the latest version.

yhonatanb20, 2022-03-21:
Thank you! I feel comfortable with generating apks and updating the app, but I have no experience using log files, debugger etc. So it is better for me not to use dev branch on phone, right?

Andries-Smit, 2022-03-21:
@yhonatanb20 you can just update the watch app and keep the releases 3.0.0.1

yhonatanb20, 2022-03-21:
So it is not really risky?

yhonatanb20, 2022-03-21:
OK. Can you please explain when do I switch to the dev branch while building the apk? Is it on the GIT -> PULL menu?

closed 2022-03-24 by MilosKozak