nightscout/AndroidAPS #1475: Minor actions tab 3 dot menu bug

robertrub, 2022-03-21:
Actions tab, I hit 3 dots menu. I see the preferences for the actions but I can't tap and open the option.

[image]

[image]

Thomas-Wolff, 2022-03-21:
Yes, I also notice the same behaviour.
In the tabs where it works, the arrow key at the top left no longer brings you back.

Greetings Thomas

robertrub, 2022-03-21:
@Andries-Smit I know that there are no options fir actions. The bug is that the "Action pref" line is in the list of the menu. Unclicable but there ;)

closed 2022-03-22 by MilosKozak