nightscout/AndroidAPS #1477: [3.0.0.1-dev-g] Notifications from automations display twice

kristinsmotz, 2022-03-21:
Have noticed since building dev-g this weekend, and it persists after rebuilding today, that notifications from automations are duplicated in the header of AndroidAPS, once with red background and once with orange. One of them disappears if the system notification is swiped closed, but the other one must be OK'ed within AndroidAPS. 
Relevant screenshots:
[image]
[image]
[image]
[image]

Will add Log when I get to my computer later.

kristinsmotz, 2022-03-22:
[AndroidAPS_LOG_1647894842341.log.zip](https://github.com/nightscout/AndroidAPS/files/8322819/AndroidAPS_LOG_1647894842341.log.zip)

MilosKozak, 2022-03-22:
you setup sending notifications to NS and receive it back too. Then you get them twice

closed 2022-03-22 by MilosKozak