nightscout/AndroidAPS #1478: Constantly Notification from ANDROIDAPS PumpControl

rdurbano, 2022-03-21:
Hello everyone!

I installed the last version yesterday and I noticed that the ANDROID APS PumpControl sending notifications for all glucose glucose information to xDrip and Nightscout. See the imagem below:
[image]
[image]

MilosKozak, 2022-03-22:
why are you doing profile switch every 5 min?

rdurbano, 2022-03-23:
Is it a option in the app ? Is it possible disable ?

duncanbarnes, 2022-04-10:
Possibly the same as #1536

> why are you doing profile switch every 5 min?

This looks the same as I see in which case it's happening by itself, not a manual action.

MilosKozak, 2022-05-27:
resolved in dev

closed 2022-05-27 by MilosKozak