nightscout/AndroidAPS #1479: Reservoir Status light in NSClient

shafiro, 2022-03-22:
NSClient 3.0.0.1-dev-g
Build 899270b7e-2022.03.20-08:35
Flavor: nsclientdebug
Nightscout version: 14.2.5
Omnipod Dash
Engineering mode enabled

Reservoir status light is always red on NSClient follower (Xiaomi mi9). Regardless of actual units remaining.
Reservoir status light is not red on main AAPS Phone, and not red in NS site. Just in NSClient follower.
It stays red even if "status lights" are turned off in NSClient preferences.
 
[image]
[image]

MilosKozak, 2022-03-22:
what pump?

shafiro, 2022-03-22:
Dash.

kristinsmotz, 2022-03-23:
Pretty sure the status light color in NSClient has been red regardless of actual remaining amount for some time, I've seen this while using Eros as well. Checked my screenshots back to July 21, and all are red even with more than 50U left

shafiro, 2022-03-24:
Yes, this goes a long way back and we had it with Eros on 2.x as well.

MilosKozak, 2023-08-17:
Closed for inactivity. Create new issue is if you still see the problem on latest branch

closed 2023-08-17 by MilosKozak

olorinmaia, 2023-08-17:
I can confirm the issue persist on latest master, if latest branch is what you mean by that. Omnipod dash.

parapenT1sta, 2023-08-17:
> I can confirm the issue persist on latest master, if latest branch is what you mean by that. Omnipod dash.

DEV branch is the latest: https://github.com/nightscout/AndroidAPS/tree/dev