nightscout/AndroidAPS #1486: NSClient - translation issues

andrzej-aps, 2022-03-23:
although in Preferences>General>Language the language is set to english there is still some translation to default system language which in my case is polish
[image]
This issue occurs on:
-hamburger menu
-home screen
-local profile screen
-three dot menu
There are also different options in preferences depending on the default system language.
When the default system language (in my case polish) is different than the one in preferences (in my case english) there is additional option for NSclient preferences -> synchronisation
[image]
When the default system language equals the language that is set in preferences then the NSClient preferences do not have option to change synchronisation
[image]

Philoul, 2022-03-24:
Did you check in crowdin if translation of missing option is correct?
It could be a wrong character use in polish translation that create this issue in aaps...

andrzej-aps, 2022-03-24:
I didn't check crowdin. But I checked with czech language. I found out that if NSclient language is set to same language as default system language then there is no option to choose synchronization options in NSclient preferences. I don't know how it is translated but I was expecting that english is the default language for NSclient.

ninelore, 2022-04-10:
This issue also happens on full AndroidAPS and in my case with German AAPS language setting