nightscout/AndroidAPS #1504: Trend arrows are displayed incorrectly.

elgrupo123, 2022-03-26:
Trend arrows are displayed incorrectly.
For a long time I have observed that the divider arrows in AAPS are displayed incorrectly on high climbs. In XDrip, the are displayed correctly and forwarded to AAPS.

d33ce02 Omnipod Dash/Libre2/Pixel 4a.
[image]
[image]

MilosKozak, 2022-03-26:
you need to provide logs with time when this happen

elgrupo123, 2022-03-26:
[AndroidAPS_LOG_1648328454277.log.zip](https://github.com/nightscout/AndroidAPS/files/8356435/AndroidAPS_LOG_1648328454277.log.zip)

MilosKozak, 2022-03-28:
I need exact time + log from this time + screenshot 
together

elgrupo123, 2022-03-28:
Here's another example.
Here's another example.
Saturday, March 27, 2022 at 2:00 p.m

[image]
[image]
[image]

[AndroidAPS_LOG_1648382552992.log.zip](https://github.com/nightscout/AndroidAPS/files/8361345/AndroidAPS_LOG_1648382552992.log.zip)

MilosKozak, 2022-04-01:
`Received xDrip data: Bundle[{com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel=200, com.eveningoutpost.dexdrip.Extras.BgSlope=8.652649374679686E-5, com.eveningoutpost.dexdrip.Extras.VersionInfo=2122032611 b66817e-2022.03.26, com.eveningoutpost.dexdrip.Extras.Raw=209.99999999999997, com.eveningoutpost.dexdrip.Extras.Time=1648382345344, com.eveningoutpost.dexdrip.Extras.CalibrationInfo=1648071747297, com.eveningoutpost.dexdrip.Extras.NsNoiseLevel=null, com.eveningoutpost.dexdrip.Extras.Noise=-9999.0, com.eveningoutpost.dexdrip.Extras.SensorBattery=-1, com.eveningoutpost.dexdrip.Extras.CalibrationPluginInfo=OOP , com.eveningoutpost.dexdrip.Extras.NoiseWarning=0, com.eveningoutpost.dexdrip.Extras.BgSlopeName=DoubleUp, com.eveningoutpost.dexdrip.Extras.BgEstimate=210.0, com.eveningoutpost.dexdrip.Extras.SourceDesc=Other App, com.eveningoutpost.dexdrip.Extras.SourceInfo=Libre2 Native, com.eveningoutpost.dexdrip.Extras.Collector.NanoStatus=}]`

`Received NS Data: [{"_id":"62405189e13c53240ae83f39","mgdl":210,"mills":1648382345344,"device":"xDrip-NSFollower","direction":"Flat","filtered":0,"unfiltered":0,"noise":1,"rssi":100,"type":"sgv","scaled":210}]`

* you receive from xDrip with correct direction
* then you receive data from NS with direction "Flat", data is marked comming from xDrip-follower

you should definitely review your configuration and:
* define single path to upload to NS
* all other clients put to read only mode
* disable download of Bg data from NS if you want to receive them from xDrip in AAPS

closed 2022-04-01 by MilosKozak

elgrupo123, 2022-04-07:
thank you very much for the advice. 
It worked!  Thanks!!!