nightscout/AndroidAPS #1505: Google Pixel 5a - AndroidAPS Randomly Stops

JimAdkins, 2022-03-27:
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
3-26-22 around 6:49pm Arizona TZ
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
3.0.0.1
- Obtain the app's log files, which can be found on the phone in
Log file attached
[AndroidAPS_LOG_1648346674124.log.zip](https://github.com/nightscout/AndroidAPS/files/8356690/AndroidAPS_LOG_1648346674124.log.zip)

My son is using a Google Pixel 5a, and it randomly shuts AndroidAPS off. I will then get an alarm saying there was a APS error, and call him to open the app again on his phone. I have ensured that all of the settings are enabled for the app to work in the background, but it still randomly shuts off. 
We are still very new to AndroidAPS (Using it about 4 weeks now,) so if there is some PEBKAC issue, apologies. 
Thanks so much in advance!

MilosKozak, 2022-03-28:
sorry, the reason is not visible from logs

closed 2022-03-28 by MilosKozak

JimAdkins, 2022-04-01:
Thanks for having a look. I appreciate your time. I just captured a log where it stopped running again. We don't get any errors on the screen. It doesn't say the app has stopped working, it just stops, and we don't know until we get an OpenAPS alarm after 15 minutes. Here is the log, and here is the exact point at which the log stops, and I started the app again. 11 minutes went by.

[AndroidAPS._2022-03-31_18-36-31_.0.log](https://github.com/nightscout/AndroidAPS/files/8394112/AndroidAPS._2022-03-31_18-36-31_.0.log)

18:36:29.675 [NSClientPluginHandler] D/EVENTS: [RxBus.send():23]: Sending EventNSClientUpdateGUI[]
18:36:31.806 [main] D/EVENTS: [RxBus.send():23]: Sending EventChargingState[batterLevel=86,isCharging=false]
18:36:31.807 [main] D/CORE: [ChargingStateReceiver.onReceive():22]: EventChargingState[batterLevel=86,isCharging=false]
18:47:36.905 [main] D/CORE: [MainApp.onCreate():76]: onCreate
18:47:36.910 [main] D/EVENTS: [RxBus.send():23]: Sending EventNewBG[glucoseValue=<null>]