nightscout/AndroidAPS #1506: DST Dana RS stuck at reading history

mxsrm, 2022-03-27:
After DST AAPS is now stuck at reading the pump history. Manually fixing the pump time (which was already updated manually) and resetting the APS databases did nothing. Even though the. Message disappears after some time,  looping, setting temp ratios Etc. does not work.

AAPS 3.0.0.1

CaptainBalou, 2022-03-27:
Same here. Can't bolus anymore. I'm sorry but i can't attach logs Herr because I'm mobile.

3.0.0.1-dev-g from three days ago

CaptainBalou, 2022-03-27:
After issuing one bolus via the bolus calculator directly in the Pump AndroidAPS seems to work fine again. Bolus history can be read without error now.

CaptainBalou, 2022-03-27:
[AndroidAPS_LOG_1648366340572.log.zip](https://github.com/nightscout/AndroidAPS/files/8357192/AndroidAPS_LOG_1648366340572.log.zip)

Finally i got the file uploaded. Hope this helps.

CaptainBalou, 2022-03-27:
Also some additional screenshots If needed...

[image]

[image]

[image]

CaptainBalou, 2022-03-27:
After the manually issued bolus via the Pump all Looks good which is like this:

[image]

As you can see also all data since DST change ist now covered in the chart.

mxsrm, 2022-03-27:
I can confirm manually bolusing on the pump helps.

npsgr, 2022-03-27:
The solution is “delete history” in DanaRS. The procedure is 
1.	go to pump
2.	Unlock
3.	Then while pressing - at the same time press + sign and arrow
4.	Appears password  0000
5.	Go to third 0 and put 4
6.	Go to fourth 0 put 9
7.	Go to first 0 put 8
8.	Go to second 0 put 9
9.	All in this order 
10.	Message then delete history.

closed 2022-03-27 by MilosKozak

pimeys, 2022-05-19:
I've had this issue a few times in the past week already. Yes, the last message here fixes it for a few days, but it comes back eventually. It's not very nice when the connection gets stuck at night, and I don't notice any issues and I get low or high and need to wake up. The DanaRS has been working perfectly for three years with AndroidAPS. I had a replacement pump with the new firmware, which also worked over a month perfectly. When I got a new smartwatch, the problems started to appear.

Does anybody know what is the root issue here? Is it something to do with the pump clock not being sync with the watch or phone?

MilosKozak, 2022-05-19:
@pimeys communication can be affected by connected watch if phone BT stack is not good. try not to use watch for some time

pimeys, 2022-05-20:
It seems to be the cause. I disabled wifi from the watch and it's been much better now with the pump...