nightscout/AndroidAPS #1508: Feature request: Wear glucose tile

yhonatanb20, 2022-03-27:
Hello. Recently @Andries-Smit recommended me to pass to dev branch on wear. So I did, and it indeed works very good! This way I bacame familiar with the new tiles, and wanted to ask for a glucose tile: I always wanted to check my BG on wear rather than display it all time, so I've switched watchfaces every time, which works pretty well on the dev branch, but it would be wonderful if there was a tile for that.
Thank You to all AAPS developers!

Andries-Smit, 2022-03-28:
@yhonatanb20 The Wear OS Tile API is limited, and we can not make everything as on the WatchFace (No charts and limited to 1 update every 20 seconds). Can you contact me trough Discord DM to discuss your thoughts.

yhonatanb20, 2022-04-04:
I would be pleased, just don't quite understand where to go here on the Discord...

closed 2026-09-23 by olorinmaia