nightscout/AndroidAPS #1509: BUG Dev-h / Both pos activate and deactivate buttons are ON

robertrub, 2022-03-27:
Since dev h, both DASH pod activate and deactivate buttons are "lit". Usually one is "lit" and the other is grayed according to the pod status. Not any more.

[image]

[image]

robertrub, 2022-03-27:
I deactivated the old pod and both were still lit. I activated a new pod and still both are lit

Thomas-Wolff, 2022-03-28:
Hello Robby, I can confirm this, I have also done this with the EROS, see #1473.

Greetings

closed 2022-03-28 by MilosKozak

Thomas-Wolff, 2022-03-28:
@MilosKozak, I have just updated AAPS again, unfortunately both buttons are still active, or will this only be adjusted again after the next change.

[image]

[image]

Greetings Thomas

Thomas-Wolff, 2022-03-28:
Thank you, done with commit: 59d6f146d.

Greetings