nightscout/AndroidAPS #1516: AndroidAPS crashes every time I enter the Omnipod tab

Cheece777, 2022-03-29:
I just upgraded to 3.0.0.1 the other day and everything seemed to work flawlessly, however suddenly whenever I turn to the Omnipod tab, AndroidAPS crashes. Every time.
I use Omnipod Eros and OrangeLink for bridge. 
Have tried restart of phone, link and ofc AndroidAPS itself. 
Tried rebuilding AndroidAPS with same result. 
Tried building another branch - DynamicISF which worked with going to the Omnipod tab without crashing, but this branch does not allow closed loop, so want to be able to use master. :) 

My logfile is attached.
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8375271/AndroidAPS.log)

parapenT1sta, 2022-03-30:
I had similar issue but with Accu-check Combo and it was solved by selecting an option in the preferences menu.

https://github.com/nightscout/AndroidAPS/issues/1404

Cheece777, 2022-03-30:
> I had similar issue but with Accu-check Combo and it was solved by selecting an option in the preferences menu.
> 
> #1404

What option did u select? I read the other issue u referenced, and tried switching on/off autosens feature, without any luck.

MilosKozak, 2022-03-30:
#1404 was related only to Combo

MilosKozak, 2022-03-30:
no crash in log. try dev branch

Cheece777, 2022-03-30:
Dev branch is working for me, so closing this issue. Will try next master, when new version comes out. Thanks MilosKozak for looking into the log.

closed 2022-03-30 by Cheece777