nightscout/AndroidAPS #1527: No and erroneous basal rate shown

robertrub, 2022-03-31:
I'm changing my G6. During the 2h warmup, I stop the loop to have the DASH deliver the basal rate.

On the screen, I just a trait with no rate; if I hit the small line, it says 0.48U/h but the DASH tab says 0.50U/h...

I'd like to see the real rate shown on the main UI.

[image]

[image]

[image]

[image]

robertrub, 2022-03-31:
And the logs...

[image]

robertrub, 2022-03-31:
There is no basal curve on rhe graph for almost 2h either...

[image]

MilosKozak, 2022-05-19:
is this still happening?

robertrub, 2022-05-20:
Not for me. I've never see it happen again. You can close this issue.

closed 2022-05-20 by MilosKozak