nightscout/AndroidAPS #1529: Unable to compile latest version, 3.0.0.1

claresloggett, 2022-04-01:
When trying to pull and compile the latest version, I get a gradle build error. I am guessing the error must be mine, since other people seem to be using v3, but all I can discover about the error message suggests it's a code error, so I'm stuck. Any help appreciated if it is in fact my error!

Android Studio version is

Android Studio Bumblebee | 2021.1.1 Patch 2
Build #AI-211.7628.21.2111.8193401, built on February 17, 2022
Runtime version: 11.0.11+0-b60-7590822 amd64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.
Linux 5.4.0-104-generic
GC: G1 Young Generation, G1 Old Generation
Memory: 1280M
Cores: 8
Registry: external.system.auto.import.disabled=true

Current Desktop: ubuntu:GNOME

The build proceeds through many tasks then apparently errors out on the task `compileFullReleaseJavaWithJavac`. The build error I get is very long - here it is truncated, can post the whole thing on request:

```
Task :app:compileFullReleaseJavaWithJavac
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
compiler message file broken: key=compiler.misc.msg.bug arguments=11.0.11, {1}, {2}, {3}, {4}, {5}, {6}, {7}
java.lang.StackOverflowError
	at jdk.compiler/com.sun.tools.javac.comp.Attr.attribTree(Attr.java:652)
	at jdk.compiler/com.sun.tools.javac.comp.Attr.visitSelect(Attr.java:3573)
	at jdk.compiler/com.sun.tools.javac.tree.JCTree$JCFieldAccess.accept(JCTree.java:2114)
	at jdk.compiler/com.sun.tools.javac.comp.Attr.attribTree(Attr.java:655)
	at jdk.compiler/com.sun.tools.javac.comp.Attr.visitSelect(Attr.java:3573)
	at jdk.compiler/com.sun.tools.javac.tree.JCTree$JCFieldAccess.accept(JCTree.java:2114)
	at jdk.compiler/com.sun.tools.javac.comp.ArgumentAttr.visitTree(ArgumentAttr.java:207)
	at jdk.compiler/com.sun.tools.javac.tree.JCTree$Visitor.visitSelect(JCTree.java:3101)
	at jdk.compiler/com.sun.tools.javac.tree.JCTree$JCFieldAccess.accept(JCTree.java:2114)
	at jdk.compiler/com.sun.tools.javac.comp.ArgumentAttr.attribArg(ArgumentAttr.java:197)
```

MilosKozak, 2022-04-01:
try uninstall/install studio or another computer

closed 2022-04-01 by MilosKozak

mdgrs, 2022-07-23:
I got this error when using JDK 11. Other sources on the internet suggest that upgrading to JDK 12 solves the issue. I can confirm that using JDK 17 (which is LTS long term support) did solve the issue. 

If you don't have a preference (or don't know how to do this), you can install this by first installing sdkman 

https://sdkman.io/install

and then 

sdk install java 17.0.3-tem (choosing the Temurin distribution as a generically good choice)