nightscout/AndroidAPS #1536: Pump Control in MDI bug upload data Treatment in Nightscout - openaps://AndroidAPS

thiagomsoares, 2022-04-02:
Using the Pump Control 3.0.0.1 version, in MDI, AndroidAPS creates a new Treatment input for each received reading (from xdrip) with the following description Using the Pump Control 3.0.0.1 version, in MDI, AndroidAPS creates a new Treatment input for each received reading (from xdrip) with the following description **openaps://AndroidAPS** + *Profile Name in cell phone* for each new reading received. The information appears in Nightscout and also in the AndroidAPS Overview graph.+ *Profile Name in cell phone* for each new reading received. The information appears in Nightscout and also in the AndroidAPS Overview graph.

thiagomsoares, 2022-04-02:
!(https://raw.githubusercontent.com/thiagomsoares/cgm-remote-monitor/d17b9831748bbefab6139ba9d2c54e681541b97a/4f2fd694-66ca-46a1-9ad9-d03cb19fdaa1.jpeg)

thiagomsoares, 2022-04-02:
[image]

duncanbarnes, 2022-04-10:
Seeing what I believe is the same thing here as well, switched to virtual pump as a temporary workaround
[image]

duncanbarnes, 2022-04-10:
Log files from my install

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8459487/AndroidAPS.log)

MilosKozak, 2022-05-20:
resolved by https://github.com/nightscout/AndroidAPS/commit/1c8cd564917c6b9bd011eabff29b66f796dab375

closed 2022-05-20 by MilosKozak