nightscout/AndroidAPS #1538: DST: Daylight Saving time change less than 3 hours ago - Closed loop disabled

dar-kod, 2022-04-02:
AndroidAPS 3.0.0.1 Build: 479bfc410-2022.03.01

On 27.03.2022 time is changed at 2:00 and is turned forward 1 hour. From that time, for about 3 hours, closed loop have been disabled. In my own case, it caused very high B.G.s. This is not the first time with DST. This must have a better solution. 

[image]

robertrub, 2022-04-03:
Loop always turns off for 3 hours on winter/summer time change for security reasons.
Personally, I don't know what the "security reasons" are...

dar-kod, 2022-04-03:
For the last winter time change I also posted a screenshot, but not here, asking that something has to be changed as these “security reasons” are not secure for me. It could be implemented that app recognizes, if closed loop has to be disabled or not.

Alamo04, 2022-04-10:
In any case, your profile settings in pump has to be as good as possible, to avoid highs/lows in any time without loop.
You can change your alarms to handle possible highs/lows on your own in that time.

Security reason is simple, avoiding double entrys.

MilosKozak, 2022-05-19:
reason is simple. Pumps cannot handle DST changes. They track time as HH:MM:SS. During DST change time is not linear and while reading pump history some entries could be doubled which would lead to wrong IOB

closed 2022-05-19 by MilosKozak

robertrub, 2022-05-19:
Hi @MilosKozak 
I totally see your point. 

I'm preaching for my "parish", on Omnipod EROS/DASH, there is no history in the pump. Maybe for pumps without this risk, the way DST is handled could be changed. 

Robert (who is NOT a religious person ;)  )