nightscout/AndroidAPS #1540: Suggestion: More alerting options

blaxer, 2022-04-02:
I know that this is an extremely esoteric request but I think it would be very helpful to have more alerting options in AAPS. One thing that sets apart most open source diabetic projects from their commercial counterparts is that they are being developed by people with diabetes that are actually using all of this technology. We have all been sound asleep only to be woken up in the middle of the night (sometimes terrified) by some completely superfluous alarm. It would be great is AAPS could potentially have the following features:

- Allow the ability to change the alert "tones" (mp3) for different events
- Allow for scheduling of quiet and or attentive times for with different alerting options like audible or vibrate for certain types of alerts such as pump or reilylink battery low
- Change alerting thresholds based on time of day, for example during the day alert me if my pump reservoir is under 50 units at 2AM only alert me if it's down below 10units

I'm sure there are others but I believe these few would make a huge improvement in terms of livability.

ideaweb, 2022-06-08:
I would like to see it as well, especially for parents.

Its a bit "enoying" if my daughter is in the kindergarden and APS is playing music, e.g. for "high BG predicted" at 160, and someone has to take off the phone for snoozing. Here is enought that the follower app notifies.

I could mute the phone of my daughter but it mutes urgent low/hypo alerts (and phone calls) as well what i do not want.

CarnageMarkus, 2022-07-06:
I have similar experience. The app makes random sounds at random times without showing any interface. Sometimes just 2 second toast. This is unbearable and crazy.

Since approximately android 8, there are standard notification channels which provide customization at system (OS) level, allowing to bypass Do Not Disturb, customize sound, customize vibration, and others.
https://developer.android.com/guide/topics/ui/notifiers/notifications
I have "use system notification" option checked, but since there are only 2 channels, it's not really helpful.
I suggest to use one channel per notification, to give users control over those alerts.