nightscout/AndroidAPS #1541: AndroidAPS seems to sleep starting at 3 am

CaptainBalou, 2022-04-02:
Hello all together,

a friend of mine has an issue regarding AndroidAPS 3.0.0.1/master.
It looks as it often (maybe always) start sleeping starting at 3 am. Background execution permission in Android is given. The log has not a lot to say instead of "there is no activity between around 3 am and the time the app was clicked in the morning".

> 03:05:33.970 [Thread-15808] D/PUMPQUEUE: [CommandTempBasalAbsolute.execute():26]: Result rate: 0.04420000000000002 durationInMinutes: 30 success: true enacted: true
03:05:34.033 [main] D/CORE: [DummyService.onStartCommand():70]: Starting DummyService with ID 4711 notification Notification(channel=AndroidAPS-Ongoing shortcut=null contentView=null vibrate=null sound=null defaults=0x0 flags=0xa color=0x00000000 category=status vis=PRIVATE semFlags=0x0 semPriority=0 semMissedCount=0)
03:05:34.073 [Thread-15808] D/PUMPQUEUE: [QueueThread.run():155]: waiting for disconnect
03:05:35.076 [Thread-15808] D/PUMPQUEUE: [QueueThread.run():155]: waiting for disconnect
03:05:35.380 [RxCachedThreadScheduler-44] D/AUTOMATION: [AutomationPlugin.processActions():212]: processActions
03:05:35.431 [main] D/CORE: [DummyService.onStartCommand():70]: Starting DummyService with ID 4711 notification Notification(channel=AndroidAPS-Ongoing shortcut=null contentView=null vibrate=null sound=null defaults=0x0 flags=0xa color=0x00000000 category=status vis=PRIVATE semFlags=0x0 semPriority=0 semMissedCount=0)
03:05:36.078 [Thread-15808] D/PUMPQUEUE: [QueueThread.run():155]: waiting for disconnect
03:05:37.080 [Thread-15808] D/PUMPQUEUE: [QueueThread.run():155]: waiting for disconnect
03:05:38.084 [Thread-15808] D/PUMPQUEUE: [QueueThread.run():155]: waiting for disconnect
03:05:39.089 [Thread-15808] D/PUMPQUEUE: [QueueThread.run():147]: queue empty. disconnect
03:05:39.093 [Thread-15808] I/PUMP: [InsightConnectionService.withdrawConnectionRequest():305]: Last connection lock released, will disconnect in 5 seconds
03:05:39.096 [Thread-15808] D/PUMPQUEUE: [QueueThread.run():151]: disconnected
03:05:39.102 [Thread-15808] D/PUMPQUEUE: [QueueThread.run():162]: thread end
03:05:44.120 [Disconnect Timer] I/PUMP: [InsightConnectionService.setState():275]: Insight state changed: DISCONNECTED
03:05:48.409 [RxCachedThreadScheduler-25] D/AUTOMATION: [AutomationPlugin.onStart$lambda-9():140]: Grabbed new BT event: EventBTChange[deviceAddress=80:6F:B0:E4:5B:7A,deviceName=PUMP33021640,state=DISCONNECT]
03:05:48.418 [RxCachedThreadScheduler-25] D/AUTOMATION: [AutomationPlugin.processActions():212]: processActions
03:06:23.115 [AutomationPluginHandler] D/AUTOMATION: [AutomationPlugin.processActions():212]: processActions
03:07:23.186 [AutomationPluginHandler] D/AUTOMATION: [AutomationPlugin.processActions():212]: processActions
03:08:23.254 [AutomationPluginHandler] D/AUTOMATION: [AutomationPlugin.processActions():212]: processActions
03:08:46.722 [main] D/CORE: [KeepAliveReceiver.onReceive():46]: KeepAlive received
03:08:46.827 [pool-16-thread-1] D/CORE: [KeepAliveReceiver$KeepAliveWorker.checkPump():143]: Last connection: 02.04.22 03:05
03:08:47.006 [pool-16-thread-1] D/CORE: [KeepAliveReceiver$KeepAliveWorker.workerDbStatus():108]: WorkManager size is 833
06:38:51.432 [main] D/CORE: [MainApp.onCreate():76]: onCreate
06:38:51.444 [main] D/CORE: [MainApp.onCreate():102]: Version: 3.0.0.1
06:38:51.444 [main] D/CORE: [MainApp.onCreate():103]: BuildVersion: 479bfc410-2022.03.22-21:33
06:38:51.444 [main] D/CORE: [MainApp.onCreate():104]: Remote: https://github.com/nightscout/AndroidAPS.git
06:38:51.446 [main] D/CORE: [PluginBase.setPluginEnabled():80]: Starting: OpenAPS SMB
06:38:51.447 [main] D/CONFIGBUILDER: [PluginStore.verifySelectionInCategories():68]: Defaulting APSInterface
06:38:51.447 [main] D/CONFIGBUILDER: [PluginStore.setFragmentVisibilities():126]: Selected interface: OpenAPS SMB
06:38:51.447 [main] D/CORE: [PluginBase.setPluginEnabled():80]: Starting: Rapid-Acting Oref
06:38:51.447 [main] D/CONFIGBUILDER: [PluginStore.verifySelectionInCategories():79]: Defaulting InsulinInterface
06:38:51.447 [main] D/CONFIGBUILDER: [PluginStore.setFragmentVisibilities():126]: Selected interface: Rapid-Acting Oref
06:38:51.447 [main] D/CORE: [PluginBase.setPluginEnabled():80]: Starting: Sensitivität Oref1
06:38:51.447 [main] D/CONFIGBUILDER: [PluginStore.verifySelectionInCategories():89]: Defaulting SensitivityInterface
06:38:51.447 [main] D/CONFIGBUILDER: [PluginStore.setFragmentVisibilities():126]: Selected interface: Sensitivität Oref1
06:38:51.447 [main] D/CORE: [PluginBase.setPluginEnabled():80]: Starting: Profil
06:38:51.448 [main] D/CONFIGBUILDER: [PluginStore.verifySelectionInCategories():99]: Defaulting ProfileInterface
06:38:51.448 [main] D/CONFIGBUILDER: [PluginStore.setFragmentVisibilities():126]: Selected interface: Profil
06:38:51.448 [main] D/CORE: [PluginBase.setPluginEnabled():80]: Starting: BYODA
06:38:51.449 [main] D/CONFIGBUILDER: [PluginStore.verifySelectionInCategories():109]: Defaulting BgInterface
06:38:51.449 [main] D/CONFIGBUILDER: [PluginStore.setFragmentVisibilities():126]: Selected interface: BYODA
06:38:51.449 [main] D/CORE: [PluginBase.setPluginEnabled():80]: Starting: Virtuelle Pumpe
06:38:51.450 [main] D/PUMP: [VirtualPumpPlugin.refreshConfiguration():377]: Pump in configuration: Generic AAPS, PumpType object: GENERIC_AAPS
06:38:51.450 [main] D/PUMP: [VirtualPumpPlugin.refreshConfiguration():379]: New pump configuration found (GENERIC_AAPS), changing from previous (null)
06:38:51.450 [main] D/CONFIGBUILDER: [PluginStore.verifySelectionInCategories():119]: Defaulting PumpInterface
06:38:51.450 [main] D/CONFIGBUILDER: [PluginStore.setFragmentVisibilities():126]: Selected interface: Virtuelle Pumpe
06:38:51.450 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.loadSettings():85]: Loading stored settings
06:38:51.450 [main] D/CORE: [PluginBase.setPluginEnabled():80]: Starting: Anhaltende Benachrichtigung

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8403461/AndroidAPS.log)

When the app is launched in the morning the chart is stuck at around 3 am. After a few seconds the chart is then ok and the calculated basals are put in place on the pump again.

Does anyone have an idea what the issue could be here? My assumption was a missing background execution permission but this is fine. I've no idea. Can anyone help out? Thanks in advance.

Regards
Marco

Additional information regarding AndroidAPS and phone:

> AndroidAPS 3.0.0.1
Build: 479bfc410-2022.03.22-21:33
Remote: https://github.com/nightscout/AndroidAPS.git
Flavor: fullrelease
Nightscout-Version: UNKNOWN

[image]
[image]

robertrub, 2022-04-03:
Already discussed multiple times on Discord. Tell him/her to stop Google Play Protect in Play Store. As it doesn't "know" AAPS, it is shutting it down.
S/he might have a cleaning app or battery optimization too but Olay Protect is the most probable cause.

CaptainBalou, 2022-04-03:
Hello @robertrub. I searched Discord for something like hang and 3 am but this search term unfortunately didn't led me to that. Thanks for giving that hint. I'll close this incident and give it a try with deactivating Google Play Protect.

closed 2022-04-03 by CaptainBalou

robertrub, 2022-04-03:
You can look in the FAQ doc attached in #wiki.
Sometimes if you don't have the exact search words, you come out with no results. We're used to the modern search engines "interpreting largely" the search words, but it is not the case in Discord ;)