nightscout/AndroidAPS #1542: WIDGET : Very cool except basal rate color

robertrub, 2022-04-03:
The WIDGET is a great addition, tnx :)

It would be better id the basal rate was not black on gray ;)

[image]

robertrub, 2022-04-03:
Taping on the Widget doesn't open AAPS :( Would be nice to have that possibility.

robertrub, 2022-04-03:
If DynISF is selected, we have a n/a when in AAPS the info is present.

[image]

[image]

robertrub, 2022-04-03:
When carbs are required, the texts are too condensed. Could the required carbs be noted on a 2nd line?

[image]

MilosKozak, 2022-04-04:
added posibility to extend widget .... are there still any issues?

robertrub, 2022-04-04:
The _Temp Basal_ is easy to see

[image]

MilosKozak, 2022-04-04:
https://github.com/nightscout/AndroidAPS/issues/1558

closed 2022-04-04 by MilosKozak