nightscout/AndroidAPS #1544: [DEV] Enhancement: Time underneath carb icon

xJoe, 2022-04-03:
The time stamp underneath the carb icon on home screen leads to an additional line which wastes space on small phones like Jelly 2 (see screenshot). Maybe there’s an option to have it optional or remove it?
Thanks for considering.
[image]

peterleimbach, 2022-04-03:
[image]

I have no time on my Jelly 2 but I can't remember to have changed anything. I use the 3.0.0.1.

MilosKozak, 2022-04-04:
disable engineering mode

peterleimbach, 2022-04-07:
@xJoe I did not realized that you meant the dev version. Now I have it too.

[image]

MilosKozak, 2022-04-09:
should i keep this open?

xJoe, 2022-04-10:
> disable engineering mode

Thought engineering mode is mandatory to run dev or do I understand something wrong?
Proposal was meant to be enhancement for next master.

Andries-Smit, 2022-04-12:
Is the time info really needed, or added for debugging purposes? Can we remove it?

MilosKozak, 2022-04-13:
recently there was an issue when COB was sometimes outdated. I added timestamp to be able verify it.
I cannot say if the problem persists or not

closed 2022-05-27 by MilosKozak