nightscout/AndroidAPS #1545: LGS after update

Koral00920, 2022-04-03:
Hi! I've been using AAPS for almost two years with no problem and done two successful updates. This time though I had a lot of problems creating the .apk file with the latest update. I finally managed to create it after two days of constant trying, I even build the app from scratch.

After I installed it, everything seemed fine, even though at first it showed the notification that an update is needed but after restarting AAPS that notification disappeared and never showed up again. Everything worked for about two days (already after 60 days since the update was available) and today morning I suddenly noticed that closed loop is no longer available to me and I've fallen back to the Low Glucose Suspend even though I have the app updated - I can see the new options added and the 'about' info shows version number 3.0.0.1.

Can anyone help me with this problem? I'm so tired fighting with it as nothing I do doesn't seem to work. I uninstalled and reinstalled AAPS few times already and still the same - closed loop for a few minutes and then LGS back again. I thought that might've happened because I imported settings from the previous version but after the last reinstall I didn't do it and it still doesn't work... I restarted my phone several timesand even reconnected my pump again with Ruffy (ACCU-CHEK Spirit Combo) as it also showed another error - that the pump driver changed. 

Please, HELP!

mick711, 2022-04-03:
Do you try to install old version and import setting to sée what happen ?

MilosKozak, 2022-04-04:
check if you have finished objectives

Koral00920, 2022-04-04:
> check if you have finished objectives
I don't but it didn't cause any problems before - the unfinished ones have been added after I completed all when first enacting the loop two years ago. I'll complete them and see if it resolves the problem.

> Do you try to install old version and import setting to sée what happen ?
No, I didn't. I didn't think it would change anything. If completing objectives doesn't help, I'll check that.

Thanks for replies, I'll let you know if it helps :)

Koral00920, 2022-04-05:
I finally managed to finish all the questions in objective 3 (took me some time because of having to wait one hour multiple times) but still nothing, all I can have is LGS at most... I will try reinstalling again but any other ideas would be very welcome!

Koral00920, 2022-04-05:
Installed the previous version - close loop seems to work.

Koral00920, 2022-04-05:
I removed all of the previous preferences files and left only the last one from the newest version. Installed the new version. Closed loop worked for a few minutes and now is back to LGS again... I'm running out of ideas.

MilosKozak, 2022-04-05:
switching to virtual pump and back should help

robertrub, 2022-04-05:
@Koral00920  You're sure that you don't have a red triangle above the graphe.

Koral00920, 2022-04-05:
> @Koral00920 You're sure that you don't have a red triangle above left of the BG graphic ?

I do. It says BG too close.

parapenT1sta, 2022-04-05:
You need to delete them.

robertrub, 2022-04-05:
@Koral00920 You can't close the loop because of the red triangle (BGs are not reliable.

This is not a bug. Chose this issue here and search "red triangle" in the Discord channel. There are plenty messages on rhe subject.

I'll add it to the FAQ tomorrow.

closed 2022-04-05 by MilosKozak

Koral00920, 2022-04-05:
Both - switching to the virtual pump and back and resolving the problem with the red triangle helped with both issues! I didn't even think that could be the issue since with the yellow one it worked just fine. Thank you for your time to help me deal with this problem :)