nightscout/AndroidAPS #1546: 3.0.0.1 dev-h SMS communicator replies are not sent

shafiro, 2022-04-03:
3.0.0.1 dev-h
Build: d3d28605d-2022.04.03-17:55
Flavor: Full Debug
Nightscout 14.2.5
Engineering mode enabled

SMS Are received by AAPS, and shows on SMS communicator.

AAPS is replying, and the replies are shown on SMS communicator, but a text message is not created and not sent by the phone.

It is possible to send the passphrase after sending the command, and action will be taken by AAPS but very uncomfortable without the verification.

This issue was also present with 3.0.0.1 dev-g

[AndroidAPS_LOG_1649016618575.log.zip](https://github.com/nightscout/AndroidAPS/files/8405484/AndroidAPS_LOG_1649016618575.log.zip)

MilosKozak, 2022-04-04:
there is no change in SMS code for some time. review permissions and https://androidaps.readthedocs.io/en/latest/Children/SMS-Commands.html?highlight=sms#sms-commands-not-working-on-samsung-phones

shafiro, 2022-04-04:
Thanks. 
I doubled checked and made sure chat features are disabled.
I am using google messages for texts, but have also tried the samsung  app, but there is no difference.
Chat features are disable in both google and samsung apps.

MilosKozak, 2022-04-09:
there is nothing visible in logs
aaps is trying to send the message. the rest is on OS and it's settings
sorry

closed 2022-04-09 by MilosKozak

cygrahampearson, 2022-04-11:
> 3.0.0.1 dev-h Build: [d3d2860](https://github.com/nightscout/AndroidAPS/commit/d3d28605d8f34824a4b4fa64c0e8f048360946f6)-2022.04.03-17:55 Flavor: Full Debug Nightscout 14.2.5 Engineering mode enabled
> 
> SMS Are received by AAPS, and shows on SMS communicator.
> 
> AAPS is replying, and the replies are shown on SMS communicator, but a text message is not created and not sent by the phone.
> 
> It is possible to send the passphrase after sending the command, and action will be taken by AAPS but very uncomfortable without the verification.
> 
> This issue was also present with 3.0.0.1 dev-g
> 
> [AndroidAPS_LOG_1649016618575.log.zip](https://github.com/nightscout/AndroidAPS/files/8405484/AndroidAPS_LOG_1649016618575.log.zip)

Have you had any luck with a solution for this?

Mine was working perfectly on @tim2000s main dynamic ISF branch and master. When I moved to tim's dev it no longer worked. It still works perfectly on master but not on NS dev. Master works on s9, note 10, unihertz atom and jelly 2 but works on none of them with dev (with all the correct settings)

shafiro, 2022-04-11:
No. I cannot find the cause for this.

cygrahampearson, 2022-04-11:
ill run some tests on my various phones and pull some logs to see if any of
the developers can see anything

shafiro, 2022-04-21:
After upgrading S21+ To Android 12, SMS Functionality is back to normal.