nightscout/AndroidAPS #1549: 3.0.0.1 dev-h - Not tracking BG and IOB properly

cygrahampearson, 2022-04-03:
Reporting bugs
--------------
- 16:30ish - April 3rd- after updating from Tim's dev to NSDev-H I wasn't able to use Dynamic ISF because "no glucose data available". BYODA is working fine and the current BG is showing up in AAPS but AAPS doesn't seem to be logging BG data or IOB data as there are no graphs at all. IOB and COB are totally wrong. Its not logging carbs that I enter. When I manually gave insulin it worked but it actually subtracted insulin from the IOB and it subtracted the wrong amount. 
- Samsung S9
- 
I re-installed Tim's Dev and Master 3 and they are both working fine. 
-
[AndroidAPS_LOG_1649021558722.log.zip](https://github.com/nightscout/AndroidAPS/files/8405581/AndroidAPS_LOG_1649021558722.log.zip)

[image]

[image]

cygrahampearson, 2022-04-03:
I just did a fresh build of Dev-H and the same issue is happening. IOB was wrong after install. It was listed at .81 U. I gave a manual bolus of 0.05U and the IOB lowered to 0.35U. I think this could be a serious issue. Master and Tim's Dev G are still working perfectly. If you want me to do a video to show the IOB going down when insulin is given let me know

[AndroidAPS_LOG_1649026172994.log.zip](https://github.com/nightscout/AndroidAPS/files/8405668/AndroidAPS_LOG_1649026172994.log.zip)
@MilosKozak @tim2000s

MilosKozak, 2022-04-04:
nothing is visible in logs

cygrahampearson, 2022-04-04:
@MilosKozak I'll rebuild and make sure to restart phone before running to make sure logs are working properly. I'm on day 3 of covid so it might take me a couple days

cygrahampearson, 2022-04-05:
@MilosKozak the issue is resiolved in Dev-I. I took a video of the issue in dev h and will send later today

closed 2022-04-09 by MilosKozak