nightscout/AndroidAPS #1555: Default temp-targets won't set from watch

yhonatanb20, 2022-04-04:
When trying to set a predefined temp-target (ex. Eating Soon) from the new temp target tile I get an error:
[image]
Setting manual temp-target from watch works OK.
**Using dev version on wear only, 3.0.0.1 master on phone. Great tiles feature in general, thanks a lot!

MilosKozak, 2022-04-04:
i believe this is already fixed in dev. can you test it?

robertrub, 2022-04-05:
Do you have the same versions on AAPS and watch (built from the same dev version )?

yhonatanb20, 2022-04-05:
No. As I said I use dev on watch only...

robertrub, 2022-04-05:
Might explain the bug. Can you try dev on both ? I can set TT from tiles on my watch.

yhonatanb20, 2022-04-05:
I would like to do so, but since I use aaps for daily use, stability is very important for me on phone...

robertrub, 2022-04-05:
Probably fixed in one of the devs. Dev-F on watch and Dev-I on phone. Set _Activity_ from the watch tiles and phone set a TT of 150. It is now working.

I think you can close this issue.

closed 2022-04-07 by MilosKozak