nightscout/AndroidAPS #1557: Esthetic bugs on version 3.0.0.1 graph

yhonatanb20, 2022-04-04:
On version 3.0.0.1 (master) I experience two bugs regarding the displayed graph:
1. Last reading displayed on top but not on graph:
[image]
[image]
2. More rarely, there is just a part of the graph displayed (the most updated part):
[image]
*Both those problems are resolved by switching the graph all the way to 24hr and back to 3hr view.

robertrub, 2022-04-05:
I'm not sure if AAPS ever showed the BG on the graph itself.

The second point has been solved in dev.

yhonatanb20, 2022-04-05:
Of course it didn't. I mean that the last point is absent on the graph: see the first screenshot for ex.: there is no red point on the graph.

robertrub, 2022-04-05:
I don't have that missing point in dev. Maybe it was solved too. Didn't remark it missing before.

[image]

MilosKozak, 2022-04-09:
hopefully resolved in dev

closed 2022-04-09 by MilosKozak