nightscout/AndroidAPS #1560: After update to 3.0.0.1 xdrip displays wrong IOB

ArthurusDent, 2022-04-04:
As the title says, after updating to 3.0.0.1 xdrip is showing incorrect IOB when before it was usually pretty close to what had been shown in 2.8.2. 

IOB shown in 3.0.0.1 seems to be accurate and I also checked the values of the additional status bar in xdrip.

On my second phone the same wrong IOB is being shown in xdrip (Nightscout Follower) but not in NSclient.

[image]
[AndroidAPS._2022-04-04_00-00-00_.0.zip](https://github.com/nightscout/AndroidAPS/files/8413051/AndroidAPS._2022-04-04_00-00-00_.0.zip)

### System
AndroidAPS 3.0.0.1 build https://github.com/nightscout/AndroidAPS/commit/479bfc41030bf876528a53ec9a5124b224ae86a0
Android 11 (LineageOS)
Medtronic 523
Dexcom G6 with BYODA

MilosKozak, 2022-04-05:
sorry no idea how xdrip calculates these values

robertrub, 2022-04-05:
@ArthurusDent Can it come from the fact that AAPS erases IOB and COB when updating from v2.8 to v3 ?

ArthurusDent, 2022-04-05:
> @ArthurusDent Can it come from the fact that AAPS erases IOB and COB when updating from v2.8 to v3 ?

Don't think so, the screenshot was taken after a week of using 3.0.0.1.

ArthurusDent, 2022-05-18:
Currently fixed in dev. Closing.

closed 2022-05-18 by ArthurusDent