nightscout/AndroidAPS #1563: WIDGET/ Esthetic/ Change background color intensity

robertrub, 2022-04-05:
If it is not too much work, it would be nice to be able to change the saturation of the background color. The big massive gray bloc looks "old" compared to other widgets (time in the below example)

[image]

[image]

osodebailar, 2022-04-05:
As a proposal I would try a black with alpha .
Than it will have a nice glass look

robertrub, 2022-04-05:
@osodebailar What it "black with alpha" ???

osodebailar, 2022-04-05:
Black with alpha is black with x% transparency.
You can set any Color with an alpha value .
If you take black with an alpha value you get a glass effect .
If you take white with alpha it’s like a milk glass effect

MilosKozak, 2022-04-05:
https://github.com/nightscout/AndroidAPS/commit/1d2ecb257c7cafe585a6070e9614cb8d998ab752

closed 2022-04-05 by MilosKozak

osodebailar, 2022-04-05:
@MilosKozak : perfectly done

MilosKozak, 2022-04-05:
setBackgroundResource doesn't have rouded corners