nightscout/AndroidAPS #1564: [3.0.0.1-dev-i] Error in BG Check vlue when transferred fro AAPS into NSClient

reijoahola-nightscout, 2022-04-05:
When giving "BG Check" value in AAPS, it is transferred also into NightScout and NSClient app but
BG value is wrong in NS and NS Client. The value in NSClient is the right BG value divided by 18.

This was tested with the latest 3.0.0.1-dev-i but the problem has been also in earlie dev releases.
I am not sure is this a problem in NS or AAPS/NSClient.

kristinsmotz, 2022-04-05:
I logged two BG checks last night/this morning. They show the correct mmol/l value in Nightscout report "Treatments" tab, but wrong value (mmol/18) in NSClient, if that helps at all. 
[image]
[image]
This log is from NSClient app. 
[AndroidAPS._2022-04-05_11-50-40_.0.zip](https://github.com/nightscout/AndroidAPS/files/8421635/AndroidAPS._2022-04-05_11-50-40_.0.zip)

MilosKozak, 2022-05-27:
I tested now (AAPS dev, NS dev) and works like expected

closed 2022-05-27 by MilosKozak