nightscout/AndroidAPS #1567: DASH / Feature : Synchronize DASH automatically to stop alerts

robertrub, 2022-04-05:
The DASH beeps when it reaches the minimum reservoir set by user, at 70 hours and 72 hours of use (as well as before 80 hours etc). When we hear these beeps, we are obliged to open AAPS, go to DASH tab, force an _actualization_ so that AAPS can show the "Accept alerts" button which permits to stop the beeps. 

All these situations are known to AAPS which keeps track of the insulin in stock, pod end time etc. 

Can AAPS _Actualize_ pod automatically when it knows that these alerts situations will happen so that the "Stop Alerts" button will be available to stop the beeps easily ?

kristinsmotz, 2022-04-05:
I raised a similar request to copy the "automatically acknowledge alerts" setting/functionality from the Eros setup to the Dash as well. i think it would be harder for the system to check/listen for the alerts than to automatically dismiss them once there is a connection for another reason/TBR/SMB/keep alive or otherwise.

nicoleloop, 2022-04-05:
It can also lead to slightly embarassing situations when out in public; when others in your immediate vicinity look at you, wondering why/how your abdomen is beeping at them...

robertrub, 2023-08-22:
This one is old but still a valid enhancement ;)

kristinsmotz, 2025-01-22:
There are two different issues her: 
Auto-acknowledge pod alerts (pod will expire, low reservoir etc, the usual bibeep, bibeep). These could be autoacknowledged on next manual or automatic communication with pod, and replaced with a careportal note (would make them visible to parents in AAPSClient as well). The Omnipod Eros setup had a preference option to auto-acknowledge these alerts, in which case the app would dismiss the alert to stop the reminders every few minutes, and we wouldn't have to scramble to the phone, into dash tab, wait for refresh and then silence alerts. 

The more serious death scream that means the pod is no longer working at all, those would be harder to automate I think, since it would require deactivating a pod. It is probably a good idea to make this visible to parents in AAPSClient by copying the error message to a Note or Notification/Alert. But automatically deactivating a screaming pod to make it stop screaming is definitely a bad idea and will inevitably lead to middle of night pod failures being ignored until morning, by which time it will be hard to avoid DKA incident which can be dangerous.