nightscout/AndroidAPS #1572: Automation before Loop

JohnKitching, 2022-04-06:
Would it be possible for automation to run and complete before the Loop calculation is done when receiving a new BG?
I'm not certain but this seems to be the chain of events:

1. New BG received
2. Loop starts calculating based on previous profile percentage
3. Automation runs and changes the percentage based on the BG received in 1.
4. Loop finishes calculating and makes changes.

It would work much better for me (and I suspect others) if 3. came before 2.

MilosKozak, 2022-04-06:
changing profile based on BG in automation is wrong approach 
for example
if (BG > 8) set profile 130% 
means that if BG is higher that 8 body needs basal +30% which is not true

JohnKitching, 2022-04-06:
It works very well for me but I'm not asking for a change related to that. I'm just asking for automation to run before the loop calculation. To me that's a logical thing to do regardless of why. 

My preference would be to just change CR and ISF (which are widely acknowledged as changing with BG) but there doesn't seem to be a way of achieving that unless I do profile switching which then resets autosens.

closed 2026-10-15 by MilosKozak