nightscout/AndroidAPS #1582: Double BG after G6 calibration is back

robertrub, 2022-04-07:
@MilosKozak You had solved rhe double BG with calibration near the 5 min after last read. This happened this morning with a G6 calibration just after rhe "normal" BG read. G6 with BYODA 1.9

[image]

[image]

[image]

[image]

[image]

MilosKozak, 2022-04-09:
this part is missing in logs. could you look for the correct one?

robertrub, 2022-04-09:
Oups sorry. This one should be the correct one.

[image]

robertrub, 2022-04-09:
Strange as I had exported the log rapidly after the bug. Sorry, it's the only one I have.

I'll try to trigger the red triangle with the next calibration and catch the correct log.

robertrub, 2022-04-14:
@MilosKozak Could not trigger the red triangle yet. Still trying to calibrate near the "normal BG read" times by the transmitter.

robertrub, 2022-05-01:
@MilosKozak I'm closing this issue. Was never able to create the situation again.

I'll open a new one if it happens again.

Rob

closed 2022-05-01 by robertrub