nightscout/AndroidAPS #1586: Feature request: Period Calendar within AAPS with cycle graph of TDD

kristinsmotz, 2022-04-07:
Question: Has there been any thought about incorporating a simplified period calendar into AndroidAPS? I'm imagining something like a button on the Actions tab to set a specific note type for first day/last day, and a separate tab with a calendar where the tagged days are marked (and a toggle to list view with count of days between each first day). After 2-3 cycles (or from preferences with an initial number) the future parts of the calendar view could have a predicted start date/duration. If possible, a graph of TDD's for each cycle so we can see variations to better predict necessary profile changes etc. Ideally we could sync to nightscout so it could also be available in the NSclient app for parents of tween/teens. I know there are several good period apps available, but I don't know of any that can talk to nightscout/dexcom or otherwise give the option for that correlation between glucose/TDD and hormonal cycles.

alzorn, 2022-04-07:
I like the idea.
Currently I always activate a profile of 110% of the regular one roughly one week before my period, when I notice that the insulin does not suffice anymore. Being able to insert the information, being reminded that it's about time and the other ideas sound really great :-)

nicoleloop, 2022-05-03:
I also support this feature request, as it would benefit pretty much every woman using AndroidAPS between the ages 13-60 years. Has anyone else noticed that this feature request was actually requested, as of yet? 😉 I'm sure that some of us women could figure out how to help create a simple once a month reminder alarm with a bit of extra logging capabilities included for the initial 2-3 month cycle long memory of said simple alarm message, if anyone is confused about how to do that?

kristinsmotz, 2022-06-29:
Well my daughter who'll be 11 in August actually got her period last week, so this is at the front of my mind again :D 
I'd like to be able to explore this myself, but I'm not even sure how hard it would be to add one or two new TherapyEvent types (Start period/end period) to get started without affecting other types of notes. For now I added a regular note that she started, and installed a separate app for cycle tracking.

nicoleloop, 2022-06-30:
I'm sure that we can figure something out.

ArthurusDent, 2022-11-30:
It's been almost eight months. I'm not seeing any discussions, requests for feedback or pull requests.  Is somebody actually working on this?

kristinsmotz, 2022-11-30:
I've tried looking into it but sadly I'm not even able to find out where to start due to having no development experience on Android apps, or time on my hands to "figure it out".

nicoleloop, 2022-11-30:
> There is an open-source period tracker developed in Berlin that is privacy aware and keeps its data on the device.
>
>
>
> Source:  https://gitlab.com/bloodyhealth/drip
>  Website:  https://bloodyhealth.gitlab.io/
>
>
>
> Maybe women would like to use just one "real" period tracker and have it share its data with AndroidAPS?
>
>
>
> AndroidAPS could then also upload to Open Humans to hopefully be able to do some big data/machine learning magic.

No offense, but I can't imagine anything that I would enjoy less than having my period timing information shared with the open human project. 🤨

ArthurusDent, 2022-11-30:
I mean, sure, let's have an option to select which data will be shared with Open Humans. But it's not like the data is being directly connected to your identity. You choose a pseudonym and your data is connected to that pseudonym. Depending on the data you provide, it could be more or less difficult to identify you personally. More information here: https://www.openhumans.org/data-use/

You could of course manage your profile changes manually or set up automations. I just thought that women might also like more algorithmic automation, if possible. Like we currently have "autosense", we would then have "periodsense" or something like that. One algorithm more to battle the chaos of hormones. 

But for that to happen, you probably need data first. And since there is already a framework for that in AAPS, it just seemed logical to add that capability and hopefully in the future someone can check whether an algorithm could even help with that problem.

nicoleloop, 2022-12-04:
My apologies ArthurusDent! My perspective is somewhat skewed by the recent changes to abortion access in the USA. Prior to that and learning that some apps for period tracking within the Google Play Store were being weaponized as spying devices to try to figure out when women stopped their monthly menses suddenly, indicating that they might be pregnant. This info was being forwarded to law enforcement in states that have made receiving abortions illegal after differing amounts of time after conception. So that was the only thing which worried me about the Open Humans integration option...that this info might de-anonymized by state or local government agencies without the knowledge of the Open Humans Project (being open-source tech) which could possibly cause problems for those donating their data to the Project itself. But perhaps a small disclaimer would take care of this issue, rather than just not pursuing this avenue at all. 🙂

ArthurusDent, 2022-12-06:
No worries! You have a valid point and I read about how problematic those apps can suddenly become when laws change but I did not connect the dots when we spoke about Open Humans, so thanks for pointing out that important aspect.

tim2000s, 2023-01-28:
Allow integration with Fit's cycle tracking to allow automations to make changes to profiles accordingly.

Potentially allows automation with, for example, Oura, Natural Cycles and Fit to learn about when cycles occur and then automatically adjust settings.

I mention Natural Cycles, as it was recommended to me for use with Ours by a friend.

ArthurusDent, 2023-01-28:
There is a new Google app called [Google Health Connect](https://play.google.com/store/apps/details?id=com.google.android.apps.healthdata&hl=de&gl=US), which collects health data from different health apps. The app is still in beta. 

Currently, it is really difficult to share health data between different vendors. What Health Connect tries to achieve is to act as a data hub for all the health data somebody generates by converting that data into the same standard. Then the user can decide which app on the device should get access to that data or just parts of it. Apps that send and read data need to be compatible, so right now not a lot of apps are compatible. The period tracker Flo is already compatible.

It looks like this solution is privacy friendly. Health Connect doesn't sync with a cloud and the data stays on the device. As long as the other collecting apps (like Flo) or apps with access to the data don't upload it, it should also stay on device. 

There are some smaller downsides of using that option: the Health Connect app uses more battery and the data sharing is not in real-time. In a German article about the app they observed a delay of up to half an hour. That's probably not a problem for period data, but it might be a problem if we were to discuss further automations, e.g. for unplanned physical activity. 

Link to the German article (paywalled): https://www.heise.de/ratgeber/Google-Health-Connect-Fitnessdaten-App-uebergreifend-synchronisieren-7467813.html?seite=all

So the idea would be to check whether AndroidAPS could be able to get access to Health Connect data. That way users would not need to use one specific AndroidAPS-compatible period tracking app but only one that is compatible with Health Connect.

szantos, 2023-02-15:
As a first iteration in AAPS, I propose a simple solution:
- a counter, which counts up +1 each day
- consists of max. 4 characters: DDTT (DD=00-99 day counter, TT=custom text)
- maybe in the status lights row?

Short press: set day - ability to (re)set the counter
Long press: set options - ability to set custom text for given days

Example for configuration: 01-05,ME ; 06-10,FO ; 11-13,FE ; 14,OV ; 15-21,L1 ; 22-28,L2
On days 1-5 "ME" is shown after the day counter, like: 01ME

(Colors could be made possible, too: 01-05,ME,red ; 06-10,FO,white ; ...)

Suggested new automation triggers:
- day counter days
- day counter text

Benefits:
- visual feedback 
- custom text for better orientation
- new automation triggers to help the management
- not only for period (menstrual cycle)

See bottom left corner:

[image]

kristinsmotz, 2023-02-16:
> As a first iteration in AAPS, I propose a simple solution:
>
>     * a counter, which counts up +1 each day
>
>     * consists of max. 4 characters: DDTT (DD=00-99 day counter, TT=custom text)
>
>     * [...]

This looks good as a first version. I'm curious if we'll be able to access the data later, maybe as a part of statistics page (or a separate page) to connect cycle day to TDD? I'm thinking that setting up automations based on the data would be much easier if we could compare cycles.

MilosKozak, 2026-10-16:
v4 supports mothly insulin review for this purpose